FreeCAD/FreeCAD #22882: [Preferences] Font in tree elements (LHS) is different from application set font, I believe this is a recent regression

Syres916, 2025-08-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Expected result (from version 1.0.1):

[image]

Actual result using three different themes with 1.1dev:

[image]

[image]

[image]

### Full version info

```shell
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42790 (Git)
Build date: 2025/08/01 20:22:01
Build type: Release
Branch: main
Hash: a83f34e31ce49415288f3c2f1fb7cc844d980935
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9567/1
Installed mods: 
  * ChangeAppFont
  * FreeCAD-themes 2025.1.7
  * Assembly4 0.50.18
  * CfdOF 1.30.4
  * Woodworking 1.0.1.39285.1.0.1.2025.06.04
  * sheetmetal 0.7.22
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Anything else?

The ChangeAppFont code used in both 1.0.1 and 1.1dev sets Sans Serif as FreeCAD font:

```
preferred_fonts = ["Segoe UI", "Sans Serif", "MS Shell Dlg 2", "Arial", "Helvetica"]
from PySide import QtGui
from PySide import __version_info__
if __version_info__[0] == 5:
    available_fonts = QtGui.QFontDatabase().families()
else:
    available_fonts = QtGui.QFontDatabase.families()
if QtGui.QApplication.font().family() == "Monospace":
    for available_font in available_fonts:
        if available_font in preferred_fonts:
            myFont = QtGui.QFont(available_font, 10)
            QtGui.QApplication.setFont(myFont)
            break
App.Console.PrintLog("Application font set to: " + str(QtGui.QApplication.font().family()) + "\n")
```

If I don't use this code the result is even worse as the Preferences comboboxes also have the same incorrect font as the tree elements.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-08-03:
Root cause is https://github.com/FreeCAD/FreeCAD/commit/fcd1b923b387d9caa9dbadc28d3cc8af03a43b9c surely if this was a MacOS specific fix the existing and new code should have been wrapped in:

```
#ifdef FC_OS_MACOSX

#else

#endif
```

@leematos ?

leematos, 2025-08-04:
@Syres916 Thanks for the ping.

I think i see how this happened. I may have overridden a typeface in the Gui namespace.

The intention is not "mac-specific" but to avoid "hard coded courier". As a side effect of that, I think I clobbered "font" here: https://github.com/FreeCAD/FreeCAD/commit/fcd1b923b387d9caa9dbadc28d3cc8af03a43b9c#diff-1793b696bf0074f420e17396d6d2264bf5d60178c8730c49415d364e5b6e48b9R857

I tried to download the latest mac build from 2 days ago but my mac says it's corrupted. 

If we need to revert feel free, I'll try and get a build env set up and see if I can recreate this on my Mac.

Syres916, 2025-09-06:
@adrianinsaval if you're happy with the hotfix, we can keep this open and after merge remove the Blocker tag?

closed 2025-10-07 by chennes