FreeCAD/FreeCAD #22887: Crash in expression editor when selecting dropdown item

eliasp, 2025-08-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting a 2nd order item from the dropdown of the expression editor, FreeCAD crashes.

How to reproduce:
- open the expression editor for any property of an object
- select an object from the auto-completion dropdown of the editor
- enter a period and a letter to trigger the further auto-completion of this object's properties, e.g. `.P`
- select one of the suggested properties, e.g. `<<Cube>>.Placement`
- FreeCAD crashes

When crashing, FreeCAD prints those lines:
```
xdg_surface#91: error 2: the surface already has a role assigned xdg_popup
The Wayland connection experienced a fatal error: Protocol error
```

### Full version info

```shell
OS: NixOS 25.11 (Xantusia) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.1.Unknown
Build type: Release
Python 3.13.4, Qt 6.9.0, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/

Since this is most likely Wayland related, output from `kinfo`:

Operating System: NixOS 25.11
KDE Plasma Version: 6.4.2
KDE Frameworks Version: 6.15.0
Qt Version: 6.9.0
Kernel Version: 6.15.4 (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen AI 5 340 w/ Radeon 840M
Memory: 32 GiB of RAM (30.6 GiB usable)
Graphics Processor: AMD Radeon Graphics
```

### Subproject(s) affected?

Expressions

### Anything else?

Screenrecording of the crash:

https://github.com/user-attachments/assets/52ed07ce-ead1-4629-a021-8b34d75c0538

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-08-02:
@eliasp, do you use keyboard or mouse to select? If you don't select from the dropdown menu, just type the text, does _FreeCAD_ crash?

If you launch _FreeCAD_ as:

```
$ QT_QPA_PLATFORM=xcb ./FreeCAD
```

does it crash too?

eliasp, 2025-08-04:
I'll try to build FreeCAD and all underlying dependencies with debug symbols (which is a bit of a challenge on NixOS 😄) to get a useful backtrace which hopefully provides a clearer picture of the potential culprit.

FEA-eng, 2025-10-18:
@eliasp Can you confirm if this issue is still relevant? I can't reproduce it with these steps on Windows.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-01-18:
@eliasp Please confirm if it's still relevant.

closed 2026-01-19 by eliasp