FreeCAD/FreeCAD #22901: Issue between Assemby and Techdraw

camelator, 2025-08-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Title:
TechDraw: Missing or Incorrect Edges for Revolved Part in Exploded Assembly View (Standard Assembly Workbench)

Description:
I am experiencing a rendering issue with TechDraw in the latest FreeCAD 1.1 dev AppImage (Ubuntu 24.04). When I create a page view of an exploded assembly (using the standard Assembly workbench), some edges of a revolved tire part are missing or incorrectly shown.

Steps to Reproduce:

    Design a basic wheel in a single file:
        Create two parts: a rim (no issues) and a tire (created by sketch + revolution).
    In a new file, create an assembly using the standard Assembly workbench:
        Import both parts.
        Apply constraints; the assembly and exploded view look correct in 3D.
    Create a TechDraw page and add a view of the exploded assembly:
        Most geometry is displayed correctly, but one or two edges of the tire are missing or incorrect.
        The overall exploded logic is preserved, but the revolved tire part has edge problems.

Expected behavior:
All edges, especially for the revolved tire, should display correctly in the TechDraw page, matching the exploded assembly.

Actual behavior:
Some edges of the tire (usually one or two) are missing or rendered incorrectly in the TechDraw page view.

Forum Search:
I searched the FreeCAD forum using phrases such as:

    "TechDraw exploded view edge missing"
    "TechDraw assembly drawing incorrect edge"
    "TechDraw revolution part edge problem"
    "TechDraw missing lines in view"
    "TechDraw exploded assembly edge artifact" I did not find any posts directly addressing this specific issue.

Additional Info:

    FreeCAD Version: 1.1 dev AppImage
    OS: Ubuntu 24.04
    Assembly workbench: Standard Assembly
    The problem only appears for the revolved tire in the exploded assembly on the TechDraw page; the rim and the rest of the assembly are displayed correctly.

Attachments:
I can provide a screenshot illustrating the problem if needed.

[image]
[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/80.9812/1
Installed mods: 
  * fasteners 0.5.39
  * ExplodedAssembly
  * MakerWorkbench 1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-08-04:
Going to need a sample FCStd file to debug this.

FEA-eng, 2025-10-18:
@camelator If it's still relevant, please share the FreeCAD file to reproduce it.

luzpaz, 2025-11-11:
@camelator please respond

camelator, 2025-11-11:
Dear Luzpaz, 
I am not sure I have the project used for this error.
But it is easy to reproduce some errors, not this one:
build a tire with a "U" and a 360 degres revolution 
build a cylinder for the center of the tire

go to assembly
put the objects like I've done.

Then, and maybe I didn't use the same procedure, 
add a simple exploded view
then go to techdraw
create a view
go back to assembly
select your exploded view
and now the exception is
17:35:44  290.676 <App> Document.cpp(3018): Unknown exception in Unnamed#View thrown
17:35:44  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'SystemError'>: <built-in method recompute of TechDraw.DrawProjGroupItem object at 0x5e17f8ee4f08> returned a result with an exception set

Done with "FreeCAD_weekly-2025.11.05-Linux-x86_64-py311.AppImage"

PaddleStroke, 2025-11-21:
[image]

I think this was fixed recently by https://github.com/FreeCAD/FreeCAD/pull/25426

closed 2025-12-06 by chennes