FreeCAD/FreeCAD #22903: Stylesheets: ellipsis buttons have only a single dot

furgo16, 2025-08-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Go to the Property Editor and select any field that is editable with an ellipsis button. Notice the ellipsis button no longer has three dots, but only one:

[image]

It's probably an issue with too narrow a width.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42804 (Git) Snap 1619
Build date: 2025/08/03 23:55:53
Build type: Release
Branch: main
Hash: 54a51921827e018f78f7320d7c5d87fb0037c4b3
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Launcher
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-04:
@MisterMakerNL FYI

Syres916, 2025-08-04:
In the current _FreeCAD.qss_ I don't see any entries for `Gui--PropertyEditor--PropertyEditor > QWidget > QWidget > QPushButton` I've not gone through the whole stylesheet to see how you might be doing it differently but in the past the following is how it was done (ignore colors):

```
/* Pushbutton style for "..." inside Placement cell which launches Placement tool */
Gui--PropertyEditor--PropertyEditor > QWidget > QWidget > QPushButton,
Gui--PropertyEditor--PropertyEditor > QWidget > PushButton {
    background-color: #232932;
    border: 1px solid #232932;
    min-width: 16px; /* reset it due to larger value on regular QPushButton, same or bigger value as regular QPushButton min-height */
    border-radius: 0px;
    margin: 0px; /* reset */
    padding: 0px; /* reset */
}

Gui--PropertyEditor--PropertyEditor > QWidget > QPushButton {
    text-align:left;
    padding-left: 2px;
}
```

Is that now no longer relevant?

MisterMakerNL, 2025-08-04:
@Syres916 Good point... Seems to still be necessary, weird only the button got removed.
I did removed some hacks with QT6 in mind.
Weird that the min-width is to big and therefore the button is too small? Wtf QT.
I'm gonna test it tomorrow.

closed 2025-08-27 by kadet1090