FreeCAD/FreeCAD #22904: Stylesheets: disabled checkboxes appear more highlighted than enabled ones

furgo16, 2025-08-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Notice how disabled checkboxes seem to be even more highlighted than enabled ones

[image]

I think there are two issues at play:

- The label of a disabled checkbox should be grayed out (i.e. in a very light gray)
- The disabled checkbox fill color should be the same or lighter than that of its background, not darker

See how this used to be done for the Classic theme in 1.0.0:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42804 (Git) Snap 1619
Build date: 2025/08/03 23:55:53
Build type: Release
Branch: main
Hash: 54a51921827e018f78f7320d7c5d87fb0037c4b3
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Launcher
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-04:
It is even worse when you start hovering some checkboxes.

maxwxyz, 2025-08-04:
@MisterMakerNL FYI

furgo16, 2025-08-04:
https://github.com/FreeCAD/FreeCAD/blob/54a51921827e018f78f7320d7c5d87fb0037c4b3/src/Gui/Stylesheets/FreeCAD.qss#L248-L250

I believe the issue is that in the theme editor and the FreeCAD Light preferences the lightening of `@TextDisabledColor` is not happening. Screenshot from the theme editor in Preferences > UI > Open theme editor:

[image]

MisterMakerNL, 2025-08-04:
Ah this needs a disabled border color.