FreeCAD/FreeCAD #22927: Part Design: "Wire is not closed" warning for rectangle- opening and closing sketch solves the warning message

ErtugrulEge, 2025-08-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!  
Sometimes  FC shows "Wire is not closed" warning but  when I check  sketch with Part_CheckGeometry  FC shows no error.In this case, when I open and close sketch with doing nothing warning disappears.

[fillet_test_2.zip](https://github.com/user-attachments/files/21599830/fillet_test_2.zip)

Related issue: https://github.com/freecad/freecad/issues/16904

https://github.com/user-attachments/assets/9022bc33-4f64-40cd-a10b-a70e5ac1a3c8

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-08-06:
what solver are you using

ErtugrulEge, 2025-08-06:
Default DogLeg solver
[image]

nazar-pc, 2025-08-09:
I have a similar issue with Part workbench. Getting "Wire not closed" with perfectly normal fully constrained sketch. Is this really PartDesign-specific?

[Goodtop-10G-switch-mount.FCStd.zip](https://github.com/user-attachments/files/21698773/Goodtop-10G-switch-mount.FCStd.zip)

```
OS: Ubuntu 24.04.3 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git) AppImage
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/83.3641/1.25
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * Curves 0.6.42
```

drwho495, 2025-08-09:
> I have a similar issue with Part workbench. Getting "Wire not closed" with perfectly normal fully constrained sketch. Is this really PartDesign-specific? [...]

you have a line over a defining external geometry line, causing it to not be closed properly

nazar-pc, 2025-08-09:
Yeah, figured that out a bit later on my own as well.

Unfortunately, it was quite hard to figure that out since the list of elements has no indication of construction geometry and I don't remember older versions of FreeCAD creating external geometry as regular. But I do remember something like this mentioned in the changelog earlier this year.

The only indication is a slight visual difference in the viewport, which gets more complicated to identify visually when external geometry overlaps with other elements of the sketch.

drwho495, 2025-08-09:
> Yeah, figured that out a bit later on my own as well.
>
> Unfortunately, it was quite hard to figure that out since the list of elements has no indication of construction geometry and I don't remember [...]

I used the sketch validate tool to find the troublesome vertexes

nazar-pc, 2025-08-09:
There were zero errors detected when I tried it 🤷
When exiting sketch editing I could see some free-standing lines though, which were caused by external geometry.

Definitely something that could be improved in FreeCAD.