FreeCAD/FreeCAD #22934: Sketcher: Dimension overshoot is very big

Roy-043, 2025-08-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The overshoot of sketch dimensions seems overly large. Reducing it would look better I think.

[image]

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MisterMakerNL, 2025-08-06:
Had similar results on windows 10, definitely looks like a bug and not intentional.

maxwxyz, 2025-08-06:
@matthiasdanner @FlachyJoe any idea?

matthiasdanner, 2025-08-06:
The overshoot is a fixed length, if you zoom out it appears to be bigger because the ratio changes.

[image]
[image]

pierreporte, 2025-08-06:
Surprisingly, it is not the same for chord length and angle:

[image]

obelisk79, 2025-08-07:
I agree with Max's last suggestion. 2x arrow width which is static so it should be easy.

maxwxyz, 2025-08-13:
what is going on with your lines and arrows @MisterMakerNL the arrows have no sharp edge but a wide end from the thicker line. Are you on HiDPI screens?