FreeCAD/FreeCAD #22943: Upgrade to 1.0.2 not working. Fails to load modules.

aslze, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Just upgraded to 1.0.2 from 1.01 (used winget, which just uses the installer). Now I can't start FreeCAD. When opening a file the screen gets all white, except for the menu bar. When selecting a work bench (e.g. PartDesign) it says "DLL load failed while importing PartDesignGui: The specified module could not be found".

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

aslze, 2025-08-06:
I uninstalled and installed again and now it works.

But it is weird that *upgrading* has this issue. Worth looking at it?

adrianinsaval, 2025-08-06:
Sounds like there are conflicts between versions so the installer should be uninstalling the previous version before installing, I will change the logic in the installer and upload a new one

semhustej, 2025-08-06:
I installed 1.0.2 without uninstalling 1.0.1 on Windows without problem. The source was installer from the FreeCAD download page.

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

Fockewulf44, 2025-08-06:
Yep, reinstalling worked. Thank you!

luzpaz, 2025-08-12:
Any way to diagnose this ?

closed 2025-11-06 by github-actions[bot]