FreeCAD/FreeCAD #22949: Freecad crashes on sketch mode

Frankkkkk, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi,
When designing a part on the sketch mode, freecad can crash when entering a new constraint (see video), with the following error:

```
$ freecad
FreeCAD 1.0.2, Libs: 1.0.2R39317 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Negative datum values are not valid for the constraint with index 11
App.getDocument('Unnamed').getObject('Sketch').setDatum(11,App.Units.Quantity('189.000000 mm'))
[destroyed object]: error 1: xdg_surface must have a role object
The Wayland connection experienced a fatal error: Protocol error
```

### Full version info

```shell
OS: Arch Linux (wayland)
Architecture: x86_64
Version: 1.0.2.39317 (Git)
Build type: Release
Branch: makepkg
Hash: b0127687c2280470190116aaf2356e0e41a488a9
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, OCC 7.9.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.7.5
```

### Subproject(s) affected?

Sketcher

### Anything else?

Video upload (mkv): [freecad_bug.zip](https://github.com/user-attachments/files/21625843/freecad_bug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Frankkkkk, 2025-08-06:
Doesn't work on main neither (f1032b47d49437b1301433526bb9e0e047a2f443)

Frankkkkk, 2025-08-13:
Hey, thanks for the follow up.

Here's another simpler screencap with freecad built from 456659d763d4

The build cmd:
```
cmake -B build \
    -D BUILD_FLAT_MESH=ON \
    -D BUILD_ENABLE_CXX_STD=C++17 \
    -D BUILD_DESIGNER_PLUGIN=ON \
    -D CMAKE_POLICY_VERSION_MINIMUM=3.5 \
    -D CMAKE_INSTALL_BINDIR=/usr/lib/freecad/bin \
    -D CMAKE_INSTALL_DATADIR=/usr/share/freecad \
    -D CMAKE_INSTALL_DATAROOTDIR=/usr/share \
    -D CMAKE_INSTALL_DOCDIR=/usr/share/freecad/doc \
    -D CMAKE_INSTALL_LIBDIR=/usr/lib/freecad/lib \
    -D CMAKE_INSTALL_PREFIX=/usr/lib/freecad \
    -D FREECAD_USE_EXTERNAL_PIVY=ON \
    -D FREECAD_USE_OCC_VARIANT='Official Version' \
    -D FREECAD_USE_QT_FILEDIALOG=ON \
    -D FREECAD_QT_VERSION=6 \
    -D INSTALL_TO_SITEPACKAGES=ON \
    -D BUILD_FEM=OFF -D BUILD_MESH=OFF -D BUILD_BIM=OFF \
    -D BUILD_INSPECTION=OFF -D BUILD_FLAT_MESH=OFF -D BUILD_OPENSCAD=OFF \
    -D BUILD_CAM=OFF -D BUILD_REVERSEENGINEERING=OFF -D BUILD_ASSEMBLY=OFF \
    -D BUILD_TECHDRAW=OFF \
    -D BUILD_MESH_PART=OFF \
    -D BUILD_DRAFT=OFF \
    -D BUILD_ROBOT=OFF -D BUILD_SPREADSHEET=OFF 
    #-D PYTHON_EXECUTABLE=/usr/bin/python 
```

And the screenshot (in mp4 this time ;-) ):

https://github.com/user-attachments/assets/370b0db9-b0a8-489b-aac9-a96097e57911

Let me know if you need anything,
Cheers

luzpaz, 2025-08-27:
@Frankkkkk can you post the full [About](https://www.freecad.org/About) info for 1.1.0-dev you tested on ?

joostn, 2025-08-28:
@luzpaz: in Frankkkkk's video you can see it crashing in 1.1.0dev as well, so I think the title is wrong.

I am experiencing crashes in other places as well, for example this one:

New document, create a simple sketch (rectangle), Pad. Then, with nothing selected, start the Fillet tool:

```
Fillet: Fillet not possible on selected shapes
The Wayland connection experienced a fatal error: Protocol error
```

So it's not sketcher related. As Frankkkkk says it appears to be the floating error popup window that's causing the crash.

If I run freecad under x11 (QT_QPA_PLATFORM=xcb) it works fine.

I've now tried another PC with an Intel GPU, same issue. So it's not GPU related. Both my machines run NixOS, Frankkk's seem to be Ubuntu.

```
OS: NixOS 25.11 (Xantusia) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.1.Unknown
Build type: Release
Python 3.13.5, Qt 6.9.1, Coin 4.0.5, Vtk 9.5.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

joostn, 2025-08-28:
If I do WAYLAND_DEBUG=1 freecad
then I see this as soon as I trigger the crash:

```
Fillet: Fillet not possible on selected shapes
[2093218.119] discarded [unknown]#176.[event 1](0 fd, 12 byte)
[2093218.139] discarded [unknown]#119.[event 1](0 fd, 12 byte)
[2093218.169] {Display Queue} wl_display#1.delete_id(165)
[2093218.181] {Display Queue} wl_display#1.delete_id(180)
[2093218.185] {Display Queue} wl_display#1.delete_id(176)
[2093218.189] {Display Queue} wl_display#1.delete_id(43)
[2093218.192] {Display Queue} wl_display#1.delete_id(181)
[2093218.196] {Display Queue} wl_display#1.delete_id(182)
[2093218.199] {Display Queue} wl_display#1.delete_id(185)
[2093218.202] {Display Queue} wl_display#1.delete_id(40)
[2093218.205] {Display Queue} wl_display#1.delete_id(119)
[2093218.208] {Display Queue} wl_display#1.delete_id(65)
[2093218.211] {Display Queue} wl_display#1.delete_id(72)
[2093361.892] {mesa egl surface queue} wl_callback#184.done(5230)
[2093361.910] {mesa egl surface queue} wl_buffer#61.release()
[2093473.584]  -> wl_surface#38.frame(new id wl_callback#184)
[2093473.637] {mesa egl surface queue}  -> wl_surface#38.attach(wl_buffer#61, 0, 0)
[2093473.644] {mesa egl surface queue}  -> wl_surface#38.damage_buffer(0, 0, 2147483647, 2147483647)
[2093473.649] {mesa egl surface queue}  -> wl_surface#38.commit()
[2093473.654] {mesa egl surface queue}  -> wl_display#1.sync(new id wl_callback#72)
[2093473.983] {Display Queue} wl_display#1.delete_id(72)
[2093479.110] {Default Queue}  -> xdg_toplevel#58.set_title("* Unnamed - FreeCAD 1.0.1")
[2093485.315] {Default Queue}  -> wl_compositor#4.create_surface(new id wl_surface#65)
[2093485.350] {Default Queue}  -> wp_fractional_scale_manager_v1#14.get_fractional_scale(new id wp_fractional_scale_v1#119, wl_surface#65)
[2093485.361] {Default Queue}  -> wp_viewporter#13.get_viewport(new id wp_viewport#40, wl_surface#65)
[2093485.491] {Default Queue}  -> wl_surface#65.attach(nil, 0, 0)
[2093485.496] {Default Queue}  -> wl_surface#65.commit()
[2093485.511] {Default Queue}  -> xdg_wm_base#3.get_xdg_surface(new id xdg_surface#185, wl_surface#65)
[2093485.528] {Default Queue}  -> wp_viewport#40.set_destination(800, 46)
[2093485.534] {Default Queue}  -> wp_viewport#40.set_destination(800, 46)
[2093485.539] {Default Queue}  -> wl_surface#65.set_buffer_transform(0)
[2093485.542] {Default Queue}  -> wl_surface#65.commit()
[2093485.545] {Default Queue}  -> wp_viewport#40.set_destination(800, 46)
[2093485.553] {Default Queue}  -> wl_compositor#4.create_region(new id wl_region#182)
[2093485.558] {Default Queue}  -> wl_region#182.add(0, 0, 800, 46)
[2093485.565] {Default Queue}  -> wl_surface#65.set_opaque_region(wl_region#182)
[2093485.578] {Default Queue}  -> wl_region#182.destroy()
[2093485.623] {Default Queue}  -> xdg_surface#185.destroy()
[2093485.630] {Default Queue}  -> wl_surface#65.attach(nil, 0, 0)
[2093485.633] {Default Queue}  -> wl_surface#65.commit()
[2093485.663] {Default Queue}  -> wl_surface#65.destroy()
[2093485.673] {Default Queue}  -> wp_viewport#40.destroy()
[2093485.676] {Default Queue}  -> wp_fractional_scale_v1#119.destroy()
[2093485.705] {Default Queue} wl_pointer#26.leave(5233, nil)
[2093485.713] {Default Queue} wl_pointer#26.frame()
[2093485.717] {Default Queue} wl_pointer#26.enter(5234, wl_surface#38, 701.99218750, 398.30468750)
[2093485.725] {Default Queue}  -> wp_cursor_shape_device_v1#27.set_shape(5234, 1)
[2093485.729] {Default Queue} wl_pointer#26.frame()
[2093485.740] {Default Queue} wl_keyboard#25.leave(5235, nil)
[2093485.743] {Default Queue} wl_keyboard#25.enter(5236, nil, array[0])
[2093485.747] {Default Queue} wl_keyboard#25.modifiers(5236, 0, 0, 16, 0)
[2093485.753] {Default Queue} zwp_text_input_v3#22.leave(nil)
[2093485.757] {Default Queue} zwp_text_input_v3#22.enter(nil)
[2093485.761] {Default Queue} wl_callback#43.done(5236)
[2093485.766] {Default Queue}  -> wl_data_offer#4278190082.destroy()
[2093485.800] {Default Queue}  -> zwp_primary_selection_offer_v1#4278190083.destroy()
[2093485.812] {Default Queue} wl_keyboard#25.leave(5237, nil)
[2093485.815] {Default Queue} wl_keyboard#25.enter(5238, wl_surface#38, array[0])
[2093485.822] {Default Queue}  -> wl_display#1.sync(new id wl_callback#43)
[2093485.827] {Default Queue} wl_keyboard#25.modifiers(5238, 0, 0, 16, 0)
[2093485.830] {Default Queue} zwp_text_input_v3#22.leave(nil)
[2093485.833] {Default Queue} zwp_text_input_v3#22.enter(wl_surface#38)
[2093485.838] {Default Queue}  -> zwp_text_input_v3#22.enable()
[2093486.611] {Display Queue} wl_display#1.delete_id(184)
[2093486.625] wl_callback#184.done(2089279)
[2093487.070] {Display Queue} wl_display#1.delete_id(182)
[2093487.077] {Display Queue} wl_display#1.error(nil, 1, "xdg_surface destroyed before constructed")
The Wayland connection experienced a fatal error: Protocol error
```

joostn, 2026-03-09:
Same here, it is solved in 1.0.2.