FreeCAD/FreeCAD #22953: launching freecad linux mint

smodis, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can not launch appimage freecad 1.0.2

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7ffa276f8090]
#1  /lib/x86_64-linux-gnu/libc.so.6(cfree+0x20) [0x7ffa2774f6f0]
#2  /tmp/.mount_FreeCAAOmJfP/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7ffa1415e68e]
#3  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7ffa2947939c]
#4  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7ffa293afe2b]
#5  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#6  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#7  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7ffa293a8ba6]
#8  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ffa293d79df]
#9  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ffa293d71f7]
#10  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7ffa293ec3be]
#11  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7ffa293a1546]
#12  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#13  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#14  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7ffa29383704]
#15  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1035dd) [0x7ffa292b15dd]
#16  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x29ef1d) [0x7ffa2944cf1d]
#17  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7ffa2944c914]
#18  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x2b54be) [0x7ffa294634be]
#19  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7ffa293a1546]
#20  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#21  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#22  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7ffa293a8ba6]
#23  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ffa293d79df]
#24  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ffa293d71f7]
#25  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7ffa293ec3be]
#26  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7ffa293a1546]
#27  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#28  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#29  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7ffa29383704]
#30  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1035dd) [0x7ffa292b15dd]
#31  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#32  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7ffa29383704]
#33  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7ffa29383380]
#34  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7ffa266b0d6a]
#35  /tmp/.mount_FreeCAAOmJfP/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7ff9ed6672f4]
#36  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7ffa2947939c]
#37  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7ffa293afe2b]
#38  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#39  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#40  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7ffa293a8ba6]
#41  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ffa293d79df]
#42  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ffa293d71f7]
#43  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7ffa293ec3be]
#44  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7ffa293a1546]
#45  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#46  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#47  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x13b) [0x7ffa2938316b]
#48  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyImport_Import+0x19b) [0x7ffa293ebe5b]
#49  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModule+0x18) [0x7ffa294683a8]
#50  0x7ffa266aaa79 in Shiboken::Module::import(char const*) from /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15+0x39
#51  /tmp/.mount_FreeCAAOmJfP/usr/lib/python3.11/site-packages/PySide2/QtGui.cpython-311-x86_64-linux-gnu.so(PyInit_QtGui+0x27) [0x7ffa0c63e807]
#52  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7ffa2947939c]
#53  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7ffa293afe2b]
#54  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7ffa2939afd8]
#55  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ffa293b93af]
#56  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7ffa293a8ba6]
#57  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ffa293d79df]
#58  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ffa293d71f7]
#59  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7ffa293ec3be]
#60  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7ffa293a1546]
[... 39 lines cut ...]
#100  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x2bced3) [0x7ffa2946aed3]
#101  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8aea) [0x7ffa29466aea]
#102  /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7ffa2945afe9]
#103  0x7ffa28a0331c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libFreeCADBase.so+0x6c
#104  0x7ffa29c0d653 in Gui::Application::runInitGuiScript() from /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libFreeCADGui.so+0x43
#105  0x7ffa29d0149f in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libFreeCADGui.so+0xef
#106  0x7ffa29d0356d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7ffa29c2fac2 in Gui::Application::runApplication() from /tmp/.mount_FreeCAAOmJfP/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_FreeCAAOmJfP/usr/bin/freecad(+0x7877) [0x55bed87d9877]
#109  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7ffa276d9083]
#110  /tmp/.mount_FreeCAAOmJfP/usr/bin/freecad(+0x7bea) [0x55bed87d9bea]

### Full version info

```shell
Freecad 1.0.2 on Linux Mint 20.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-07:
@adrianinsaval FYI

Syres916, 2025-08-07:
Due to the age of Mint 20.1 (and end of support) then @smodis will have to install a more recent version of _AppImage Launcher_ IMHO. The commands to do so in the Terminal are:

```
wget https://github.com/TheAssassin/AppImageLauncher/releases/download/v2.2.0/appimagelauncher_2.2.0-travis995.0f91801.bionic_amd64.deb -O /tmp/appimagelauncher.deb
sudo dpkg -i /tmp/appimagelauncher.deb
```

This has worked for a Mint 21.x user in the past.

pyro9, 2025-08-07:
I seem to have the same problem on Ubuntu 18.04.6 LTS
both 1.0.1 and 1.0.2 have the same crash .
FreeCAD_weekly-builds-42837-conda-Linux-x86_64-py311.AppImage runs fine.

I am running them from the command line and don't even have appimagelauncher installed.

Just in case:
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3ef10) [0x7f6e30c7df10]
#1  /lib/x86_64-linux-gnu/libc.so.6(cfree+0x3d) [0x7f6e30cd694d]
#2  /tmp/.mount_FreeCAkmfHeK/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7f6e2004168e]
#3  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7f6e32bb939c]
#4  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7f6e32aefe2b]
#5  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f6e32adafd8]
#6  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#7  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7f6e32ae8ba6]
#8  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f6e32b179df]
#9  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f6e32b171f7]
#10  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7f6e32b2c3be]
#11  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f6e32ae1546]
#12  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f6e32adafd8]
#13  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#14  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7f6e32ac3704]
#15  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1035dd) [0x7f6e329f15dd]
#16  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x29ef1d) [0x7f6e32b8cf1d]
#17  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f6e32b8c914]
#18  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x2b54be) [0x7f6e32ba34be]
#19  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f6e32ae1546]
#20  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f6e32adafd8]
#21  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#22  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7f6e32ae8ba6]
#23  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f6e32b179df]
#24  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f6e32b171f7]
#25  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7f6e32b2c3be]
#26  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f6e32ae1546]
#27  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f6e32adafd8]
#28  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#29  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7f6e32ac3704]
#30  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1035dd) [0x7f6e329f15dd]
#31  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#32  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7f6e32ac3704]
#33  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7f6e32ac3380]
#34  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7f6e2f9e5d6a]
#35  /tmp/.mount_FreeCAkmfHeK/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7f6e183772f4]
#36  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7f6e32bb939c]
#37  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7f6e32aefe2b]
#38  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f6e32adafd8]
#39  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#40  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7f6e32ae8ba6]
#41  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f6e32b179df]
#42  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f6e32b171f7]
#43  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7f6e32b2c3be]
#44  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f6e32ae1546]
#45  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f6e32adafd8]
#46  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f6e32af93af]
#47  /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x13b) [0x7f6e32ac316b]
[... cut ...]
#105  0x7f6e3344149f in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libFreeCADGui.so+0xef
#106  0x7f6e3344356d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7f6e3336fac2 in Gui::Application::runApplication() from /tmp/.mount_FreeCAkmfHeK/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_FreeCAkmfHeK/usr/bin/freecad(+0x7877) [0x560d34be5877]
#109  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f6e30c60c87]
#110  /tmp/.mount_FreeCAkmfHeK/usr/bin/freecad(+0x7bea) [0x560d34be5bea]

xtemp09, 2025-08-08:
@pyro9, _Ubuntu_ 18.04 and 20.04 are no longer supported.

luzpaz, 2025-09-18:
@pyro9 i think you meant to respond to xtemp09

Nevertheless, Are you running 1.0.0 from the Ubuntu package repository or are you using the appimage that we build ? Please add your full About info so we can see.

pyro9, 2025-09-18:
@xtemp09 , I assure you, I am well aware of the Ubuntu lifecycle, including the real lifecycle as seen in in the wild. Ubuntu Pro exists exactly because some organizations are willing to pay money NOT to do-release-upgrade for an additional 5 years.

Since I am talking about the  Appimage I download and run, Ubuntu not updating the package repo other than security fixes is not really relevant to the situation. I apologize if there was confusion, but I thought this whole issue referenced the Appimage.

If the compatibility break was necessary for new capabilities in the application, I would just let it go, but since the development weeklies continue to work fine on the same machine, that is evidently not the case.

hongxima, 2025-10-25:
Hit the same issue in Ubuntu 20.04.  It happens very often (not always) when starting in terminal for its GUI mode.  However, it never happen in its --console mode.