FreeCAD/FreeCAD #22954: Sketcher constraint regression

edwilliams16, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

issue 22808 may be related
Create two points in sketcher, apply a vertical constraint between them. Then try to apply a vertical distance constraint between them.  The following error occurs
`11:18:52  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Invalid constraint index: 1
11:18:52  Unnamed#Sketch: App.getDocument('Unnamed1').getObject('Sketch').setDatum(1,App.Units.Quantity('13.213200 mm'))`

There is a thread in the forum on this [https://forum.freecad.org/viewtopic.php?t=98874](url) where the offending change has been  @theo-vt  bisected

### Full version info

```shell
OS: macOS 14.4.1
Architecture: arm64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * MeshRemodel 1.10.37
  * Vars 0.0.2.beta2
  * DynamicData 2.74.0
  * lattice2.backup1748901771.133523 1.0.0 (Disabled)
  * OpenTheme 2025.5.20
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Manipulator 1.6.0
  * fasteners 0.5.38
  * Curves_old 0.6.64
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.7.24
  * toSketch 1.0.1 (Disabled)
  * Help 1.0.3
  * Curves 0.6.64
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-08-07:
On selecting the two vertices (top one first) the Python Console output is clearly incorrect, expected Python Console output (from version 1.0.2):

```
### Begin command Sketcher_ConstrainDistanceY
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('DistanceY',1,1,0,1,26.518257))
App.getDocument('Unnamed').getObject('Sketch').setDatum(2,App.Units.Quantity('20.000000 mm'))
App.ActiveDocument.recompute()
App.ActiveDocument.recompute()
### End command Sketcher_ConstrainDistanceY
# Gui.Selection.clearSelection()
```

Actual Python Console Output:

```
### Begin command Sketcher_ConstrainDistanceY
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('DistanceY',1,1,0,1,33.134346))
ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
lastGeoId = len(ActiveSketch.Geometry)
geoList = []
geoList.append(Part.Point(App.Vector(27.072065, 20.000000, 0.000000)))
geoList.append(Part.Point(App.Vector(27.072065, 0.000000, 0.000000)))
App.getDocument('Unnamed').getObject('Sketch').addGeometry(geoList,False)
del geoList
constraintList = []
constraintList.append(Sketcher.Constraint('Vertical', 2, 1, 1, 1))
constraintList.append(Sketcher.Constraint('PointOnObject', 3, 1, -1))
constraintList.append(Sketcher.Constraint('DistanceY', 3, 1, 2, 1, 20.000000))
App.getDocument('Unnamed').getObject('Sketch').addConstraint(constraintList)
del constraintList
App.getDocument('Unnamed').getObject('Sketch').delGeometries([0,1])
App.getDocument('Unnamed').getObject('Sketch').setGeometryIds([(0,1),(1,2)])
App.getDocument('Unnamed').getObject('Sketch').setDatum(2,App.Units.Quantity('20.000000 mm'))
### End command Sketcher_ConstrainDistanceY
# Gui.Selection.clearSelection()
```

theo-vt, 2025-08-10:
I think it is, I have a fix that solves both, but there is a rendering glitch that I need to sort out

Syres916, 2025-08-11:
Maybe some context would be of use, I'm one of the users who can't move off 0.21.x due to a significant number of regressions but also the performance impact of TNP so I'm very conscious of any more impacts piled on top.

PR #23024 (2025-09-07): Sketcher: fix invalid constraint on first dimension

This PR deletes constraints before adding them back in sketcher's scale tool which is used by the autoscale feature. It also adds a way to skip solve on deletion of geometry/constraints which was necessary to fix a rendering bug. To do so a new enum DeleteOption has been created for deletion functions of SketchObject and some additional parameters have been exposed to the python API

## Issues

Fixes: #22954
Fixes: #22808
Fixes: #22248 
Fixes: #22247
closed 2025-09-07 by kadet1090