FreeCAD/FreeCAD #22962: BIM. The length of the elements differs greatly from those specified in the fields at the time of creation.

kaiwas, 2025-08-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

A video demonstration is worth a thousand words. How sad it all is.

https://github.com/user-attachments/assets/fd7c4b7d-44c8-4e6d-a1f1-2751e6c56d4e

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42837 (Git) AppImage
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Curves.backup1751366017.4487264 (Disabled)
  * addFC-main 1.6.2
  * Reinforcement
  * Curves.backup1751365993.7542331 0.6.63 (Disabled)
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * FrameForge 0.1.3
  * Curves.backup1751366131.5540717 (Disabled)
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.64
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-08-07:
Reaction to the video:
If you have `Relative` turned off, the `X, Y, Z` and `Length + Angle` show absolute coordinate values. That means either cartesian or polar coordinates of the second point as **measured from the coordinate system origin**. So it has really nothing to to with the length of the resulting wall.

I don't see any problem here. Everything that happens in the video makes sense to me.

semhustej, 2025-08-07:
> 
> [image]

Yes, this one does not make sense to me as the `Length` in the `Wall Options` panel should be 3000mm when selecting the second point and `Relative` is turned on. But I am able to reproduce this only when I am in the middle of specifying the second point. When I am entering the point coordinates I am not really looking into the wall `Length` value, so I don't find this too disturbing.

This is a different issue that what is originally discussed though.

maxwxyz, 2025-08-12:
So what is the actual issue?

kaiwas, 2025-08-12:
In my opinion, it shouldn't be like this and it looks very strange and unintuitive.
But I can't force this issue to be left open.
In my opinion, the issue is relevant, but I will understand if it is closed just to reduce the number of open issues.

kaiwas, 2025-10-19:
relevant