FreeCAD/FreeCAD #22978: BIM: Saving/Exporting IFC file ignores the IFC schema version setting in FreeCAD

semhustej, 2025-08-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When exporting (using the `Export` tool) or saving native IFC project (using `Save` tool), IFC schema setting in the user preferences is ignored. Every project is saved as 4.0 version even though 2.3 schema version is selected in the properties.
**User has no option to change the save/export version of the IFC.**

Steps to reproduce:

- go to Edit -> Preferences -> BIM -> General settings -> IFC version and switch it to `2X3`

[image]

- restart FreeCAD
- create a new project and create a wall in this project
- select the wall
- export as an IFC file with the default setting
- the exported IFC has IFC schema version 4 set. It is confirmed by opening the file in Bonsai BIM, BIMvision and looking at the text in the file

Sample FreeCAD file with the exported IFC: 

[IFC_2X3_file.zip](https://github.com/user-attachments/files/21680696/IFC_2X3_file.zip)

I am able to reproduce the issue in FC 1.0.2

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
```

### Subproject(s) affected?

BIM

### Anything else?

I am not sure whether the problem is only the label or also the IFC file structure.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-08-11:
I think the current state of the schema setting in FreeCAD is very simplified. It should be a per-project setting (or per-export setting), but there is only one switch in FreeCAD in the preferences.

When I am now thinking about it, it seems to me the export to IFC schema version 4 all the time was a conscious decision - to simplify the process. When 2x3 is selected in FreeCAD, export to 4 is easy. The other way around would be more difficult.

The case you describe - switching IFC schema mid-project - is not a typical workflow and I think it is not needed to cover it at the current state of IFC implementation in FreeCAD. I guess the export will throw errors if user creates FreeCAD objects with IFC4 setting and than switches to 2x3, but I think most users will just set the setting once and than use it.

I think what could be done is set the IFC schema version that is in the preferences to the exported IFC file when Std_Export is used.

Syres916, 2025-08-14:
I'm going to take a fresh look at the weekend, real life keeps interrupting my thought process just at the moment.

semhustej, 2026-02-21:
The issue is still valid.

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.2.0dev.20260210 (Git shallow) AppImage
Build date: 2026/02/10 18:10:36
Build type: Release
Branch: (HEAD detached at ecbe21ca0)
Hash: ecbe21ca03a59723c90ce5a677ae7ccea06ebc12
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Ireland (en_IE)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```

FEA-eng, 2026-08-31:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```