FreeCAD/FreeCAD #22989: Included Libraries are not shown in the About dialog

maxwxyz, 2025-08-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the about dialog there are no libraries shown:
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-08-08:
Works fine using self-compiled Linux build:

[image]

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42884 (Git)
Build date: 2025/08/08 11:51:18
Build type: Release
Branch: main
Hash: 32ddaff649f6c929ad25b95137ec511bc874ac43
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.2, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.2032/1
Installed mods: 
  * FreeCAD-themes 2025.6.4
  * FreeCAD-Reporting
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * CfdOF 1.30.4
  * Curves 0.6.64
  * sheetmetal 0.7.23

```

maxwxyz, 2025-08-09:
@oursland is this an issue with the weekly build mechanism?

ErtugrulEge, 2025-08-10:
Are the Version: 1.1.0dev.14555 (Git shallow)  and Version: 1.1.0dev.42884 (Git) the same thing? freecad.org->download->weekly builds redirects to this page https://github.com/FreeCAD/FreeCAD/releases  and this is not the correct page I think. 

Weekly builds with 42......   are in here : https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds 

someone changed web site link but I don2t know it is intentional or not.

JohnOCFII, 2025-08-10:
> someone changed web site link but I don't know it is intentional or not.

I know the intent is to switch the link for weekly downloads "soon" but did not realize there had been agreement to do that yet.  I'd think it'd warrant announcement on both Discord and the Forum when that happens.

Syres916, 2025-08-11:
Surely the most important issue is whether the Windows build in particular is fit for purpose. Only on Friday a user logged an issue on the [forum](https://forum.freecad.org/viewtopic.php?t=98947) with the Windows build from the new link and as soon as they followed the advice to use the one from the FreeCAD-Bundle repository the issue vanished, that to me is the _crucial_ point. All we're going to do is generate a load of false positives causing wasted time tracking them down.

PR #24118 (2025-09-30): Build: Don't remove the doc directory
It contains the ThirdPartyLibraries.html file. Fixes #22989.