FreeCAD/FreeCAD #22992: UI/UX:  Part WB Sweep tool->Sweep path guide text is not responsive

ErtugrulEge, 2025-08-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!

"Select one or more connected edges in the 3D view and press Done" text is not responsive according to task bar

https://github.com/user-attachments/assets/3bda8523-331a-4f21-ad41-e237b68c2e44

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-12:
Is this also happening in the stable 1.0 or is this a regression?

ErtugrulEge, 2025-08-12:
> Is it responsive when clicking Done?

yes. when you click Sweep path while selecting causing problem when you click done everyting returns to normal in both stable and dev versions

FEA-eng, 2025-10-19:
Isn't it supposed to stay like this when awaiting 3D view selection?

ErtugrulEge, 2026-01-19:
still relevant

ErtugrulEge, 2026-04-21:
Simple issue but still relevant
```

OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260415 (Git shallow)
Build date: 2026/04/15 00:45:21
Build type: Release
Branch: (HEAD detached at 36786d479)
Hash: 36786d47949a449b3a8585354ef8c75a1c67c03b
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR) [ OS: Turkish/Türkiye (tr_TR) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods:
  * Curves workbench 0.6.70
```

ErtugrulEge, 2026-10-21:
Small issue but still relevant 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 597.06/NVIDIA Corporation/Quadro RTX 4000/PCIe/SSE2
Installed mods:
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.81
```