FreeCAD/FreeCAD #22993: Workbench mod folder capitalization issue

Beep6581, 2025-08-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## TL;DR

The same workbench might use different capitalization in the `.local/share/FreeCAD/Mod` folder name, leading to FreeCAD thinking an installed workbench is not installed in Linux.

This is a manifestation of #22718 

## Full story

I installed the Woodworking workbench about a year ago, and it showed up in the FreeCAD workbenches list and worked correctly. However, by looking at images on https://github.com/dprojects/Woodworking today I noticed that it seemed to be missing a button (`magicStart`). I fired up the Addon Manager, clicked Refresh Local Cache, clicked Update All (I do this regularly), restarted FreeCAD, button still missing. Fired up the Addon Manager again, I went into the Woodworking entry, and FreeCAD showed that I can install it - as far as FreeCAD knew it was not currently installed, even though it was plainly available in the Workbenches list and worked correctly (albeit missing button, so old version). Strange. I tried to install it, but it timed out with `fatal: early EOF`, because my internet connection is flaky. I looked inside `.local/share/FreeCAD/Mod` and saw a `woodworking` folder, along with two `Woodworking <something>` folders, where "something" was something like "backup date". I'm not sure because I `rm -rvf ~/.local/share/FreeCAD/Mod/wood*` before I decided to file a bug report. But note the capitalization difference - `Woodworking` vs `woodworking`.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git)
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.39
  * CurvedShapes 1.0.13
  * woodworking 1.0.1.39285.1.1.1.2025.08.08
  * Curves
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Beep6581, 2025-08-14:
@maxwxyz sure.

FreeCAD shows that Addon Manager is not installed.

```
tree -d -L 2 ~/.local/share/FreeCAD/Mod/

/home/morgan/.local/share/FreeCAD/Mod/
├── A2plus
│   ├── GuiA2p
│   ├── icons
│   ├── __pycache__
│   └── translations
├── CurvedShapes
│   ├── Examples
│   ├── __pycache__
│   └── Resources
├── Curves
├── fasteners
│   ├── Doc
│   ├── FsData
│   ├── FsFunctions
│   ├── Icons
│   ├── __pycache__
│   ├── Resources
│   └── tests
├── OpenDark
│   ├── OpenDark
│   ├── QtNormalizer
│   └── resources
└── woodworking
    ├── Docs
    ├── Examples
    ├── Icons
    ├── __pycache__
    ├── Screenshots
    ├── Tests
    ├── Tools
    └── translations

32 directories
```

I run FreeCAD, update addons, refresh local cache, restart FreeCAD, and install Addon Manager. It fails with:
```
Failed to install AddonManager
Git returned a non-zero exit status: -1
Called with: /usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/AddonManager /home/morgan/.local/share/FreeCAD/Mod/AddonManager

Returned stderr:
Cloning into '/home/morgan/.local/share/FreeCAD/Mod/AddonManager'...
fatal: early EOF
```

It created an empty `AddonManager` folder:
```
tree -d -L 2 ~/.local/share/FreeCAD/Mod/

/home/morgan/.local/share/FreeCAD/Mod/
├── A2plus
│   ├── GuiA2p
│   ├── icons
│   ├── __pycache__
│   └── translations
├── AddonManager
├── CurvedShapes
│   ├── Examples
│   ├── __pycache__
│   └── Resources
├── Curves
├── fasteners
│   ├── Doc
│   ├── FsData
│   ├── FsFunctions
│   ├── Icons
│   ├── __pycache__
│   ├── Resources
│   └── tests
├── OpenDark
│   ├── OpenDark
│   ├── QtNormalizer
│   └── resources
└── woodworking
    ├── Docs
    ├── Examples
    ├── Icons
    ├── __pycache__
    ├── Screenshots
    ├── Tests
    ├── Tools
    └── translations

33 directories
```

maxwxyz, 2025-08-15:
@chennes is this a FreeCAD or Addon Manager issue?

chennes, 2025-08-15:
Sort of an Addon Manager problem, but sort of because of an inconsistency in the .gitmodules file...
https://github.com/FreeCAD/FreeCAD-addons/blob/5446e88ba0a0b5c3988016f704466cf2ce92e7c8/.gitmodules#L632

So this is effectively fixed in the latest version of the Addon Manager which is much more rigorous in its use of the addon's catalog entry as the "canonical" folder name (which is in this case with a lowercase "W"). @Beep6581 I suggest first going to the Addon Manager's preferences and disabling "git" (which is the real source of your early EOF problem), then updating the Addon Manager itself (just use the Addon Manager to install the Addon Manager... dumb, I know, but I can't retcon it!), manually removing whatever copy of woodworking you have now, and reinstalling it with the new Addon Manager. That should clear everything up and get you back up and running with the latest version.