FreeCAD/FreeCAD #22995: BIM_Setup dialogue has different set of licenses than Preferences dialogue

semhustej, 2025-08-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Priority: low

`BIM_Setup` dialog has only limited number of licenses which are available in Preferences. The licenses are also labeled different way (`Creative Common Attribution-ShareAlike 4.0` vs. CC-BY-SA).

`BIM_Setup` licenses

[image]

`Preferences` licenses:

[image]

- Not using the same license label is inconsistent and can lead to user confusion (am a really changing the same property?)
- It is not possible to select all possible licenses form the `BIM_Setup` dialogue, leading user to possibly setting up the option in two places

I am not sure if there is any reason behind having limited choice in the `BIM Setup` dialogue. If the reason is not to overwhelm the user, I think it is counterproductive because it leads to inconsistency.

**Proposed solution**
`BIM_Setup` dialogue should have the same list (including the same labels) of available licenses as `Preferences` dialogue.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-08-09:
For me there should be **one** data file to hold all licenses for the various parts of the application, currently the _AddonManager_ has IMHO the most comprehensive list in https://github.com/FreeCAD/AddonManager/blob/34d433a02c7ec5c73bec9c57d0a27ea70b36c90d/Resources/licenses/spdx.json and that way all the code whether it be C++ or Python references it, as well as the user has the ability to add/remove license(s) locally if they feel the need.

The hard coded array of licenses for the Preferences/Document list in https://github.com/FreeCAD/FreeCAD/blob/main/src/App/License.h#L46-L63 was last updated in March 2023

semhustej, 2025-08-09:
> For me there should be **one** data file to hold all licenses for the various parts of the application

That sounds reasonable to me. I don't understand coding, but from a layman's perspective this would be easier to maintain.

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/3fe35431d0201a7d985e1014a414a56feafd4551

BIM Setup gets the same license choices and labels as preferences, and keeps the matching URL.