FreeCAD/FreeCAD #23002: Assembly joints breaking even if the Face number is not changed

kwahoo2, 2025-08-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,

This does not seem to be clearly TNP issue.

I change a single dimension, no faces/edges are created/removed. See the video, the reference is "Face20" before and after the change, but still the joint breaks.

https://github.com/user-attachments/assets/0968e722-6943-471d-becf-951a9c8b1d8c

[reference-bug.zip](https://github.com/user-attachments/files/21701171/reference-bug.zip)

### Full version info

```shell
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git) AppImage
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * OpticsWorkbench 1.0.26
  * fasteners 0.5.35
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.95
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-09:
@drwho495 FYI

drwho495, 2025-08-09:
it seems that there could be two issues here: no complex find, and the part design bodies not copying element maps with allow compound disabled.

here is a video of me trying to reproduce the issue on my fork:
https://github.com/user-attachments/assets/1377d56c-f46f-4551-b6b8-c92625e4cd49

you cannot fix the first issue without using my fork, but you can fix the second issue
to do this, you need to disable the hashers on all of your files, which can be done by selecting the document on the tree view, then going into the property view panel and unselecting "Use String Hasher"

this will slightly increase file sizes, and will decrease computation time.

kwahoo2, 2025-08-09:
With string hasher disabled the reference does not break anymore.

Placement is incorrect after this, but this seems to be another, independent issue (note: Distance007 has to be changed slightly to trigger recompute):

[image]

drwho495, 2025-08-09:
> With string hasher disabled the reference does not break anymore.
> 
> Placement is incorrect after this, but this seems to be another, independent issue (note: Distance007 has to be changed slightly to trigger recompute):
> 
> [image]

Hmmm ok, could you try to independently check each joint to see if they are pointing to the right elements?

MisterMakerNL, 2025-08-12:
I have assemblies breaking out of no-where, just ad an extra mate and poof everything has lost their reference.
```
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint003.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round006 dodecahedron_half_round006.;#40199;:M;CUT;:Hd99:7,F;:Hd9b,F.Face13
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint003.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round006 dodecahedron_half_round006.;#40199;:M;CUT;:Hd99:7,F;:Hd9b,F.Face13
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Cylindrical: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint004.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round006 dodecahedron_half_round006.;#401b1:1;:M;RFI;:Hd99:7,F;:Hd9b,F.Face1
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Distance006: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint005.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round002 dodecahedron_half_round002.;#401a2:7;:M;CUT;:Hd99:7,F;:Hd9b,F.Face7
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint005.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round002 dodecahedron_half_round002.;#401a2:7;:M;CUT;:Hd99:7,F;:Hd9b,F.Face7
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Cylindrical002: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint006.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round002 dodecahedron_half_round002.;#401b1:1;:M;RFI;:Hd99:7,F;:Hd9b,F.Face1
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Distance008: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint009.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round003 dodecahedron_half_round003.;#401b1:1;:M;RFI;:Hd99:7,F;:Hd9b,F.Face1
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Distance010: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint008.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round003 dodecahedron_half_round003.;#4019c:1;:G#4019a;CUT;:H-d9b:d,E;:Hd9b,E.Edge32
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint008.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round003 dodecahedron_half_round003.;#4019c:1;:G#4019a;CUT;:H-d9b:d,E;:Hd9b,E.Edge32
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Cylindrical003: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint010.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round004 dodecahedron_half_round004.;#401a2:7;:M;CUT;:Hd99:7,F;:Hd9b,F.Face7
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint010.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round004 dodecahedron_half_round004.;#401a2:7;:M;CUT;:Hd99:7,F;:Hd9b,F.Face7
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Cylindrical004: Broken link in: Reference1
17:30:41  <PropertyLinks> PropertyLinks.cpp(494): round_dodecahedron#Joint011.Reference1 missing element reference round_dodecahedron#dodecahedron_half_round004 dodecahedron_half_round004.;#401b1:1;:M;RFI;:Hd99:7,F;:Hd9b,F.Face1
17:30:41  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Distance011: Broken link in: Reference1
17:31:04  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Distance: Broken link in: Reference1
17:31:04  pyException: Traceback (most recent call last):
  File "C:\Program Files (x86)\Freecad\Mod\Assembly\JointObject.py", line 641, in execute
    raise Exception(errStr + "Reference1")
<class 'Exception'>: Cylindrical: Broken link in: Reference1
17:31:04  pyException: Traceback (most recent call last):
[... cut ...]
<class 'Exception'>: Distance010: Broken link in: Reference1
17:34:21  Distance: Distance: Broken link in: Reference1
17:34:21  Cylindrical: Cylindrical: Broken link in: Reference1
17:34:21  Distance006: Distance006: Broken link in: Reference1
17:34:21  Cylindrical002: Cylindrical002: Broken link in: Reference1
17:34:21  Distance008: Distance008: Broken link in: Reference1
17:34:21  Cylindrical003: Cylindrical003: Broken link in: Reference1
17:34:21  Distance010: Distance010: Broken link in: Reference1

```

drwho495, 2025-08-12:
> I have assemblies breaking out of no-where, just ad an extra mate and poof everything has lost their reference.
>
> ```
> 17:30:41  <PropertyLinks> PropertyLinks.cpp(494): [...]

what workbench are you designing it in?

drwho495, 2025-08-12:
> Part design

ok good, is allow compound enabled on any of your parts?
enabling it could help fix the issue, but not fully. (the propertylinks pr needs to be merged for stuff to propogate properly)

if you dont want to enable that, then you can disablethe string hasher on all your documents for your project

or you could build my fork and use that for projects needed all of the tpn fixes i have so far

tobiasfalk, 2025-08-24:
I can confirm that there is a Problem with Assemblies if one changes a dimension.

latest nigly:

How it should be

[image]

After changing a Parameter(The screw no longer aline with the intended wholes and the joints are broken):

[image]

`12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_470 AEKS8_470.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_470 AEKS8_470.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint001.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_471 AEKS8_471.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint001.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_471 AEKS8_471.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint002.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_472 AEKS8_472.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint002.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_472 AEKS8_472.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint003.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_473 AEKS8_473.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint003.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_473 AEKS8_473.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint004.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_474 AEKS8_474.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint004.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_474 AEKS8_474.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint005.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_475 AEKS8_475.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint005.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_475 AEKS8_475.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint006.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_476 AEKS8_476.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint006.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_476 AEKS8_476.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint007.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_477 AEKS8_477.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint007.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_477 AEKS8_477.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint008.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_478 AEKS8_478.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint008.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_478 AEKS8_478.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint009.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_479 AEKS8_479.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint009.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_479 AEKS8_479.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint010.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_480 AEKS8_480.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint010.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_480 AEKS8_480.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint011.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_481 AEKS8_481.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint011.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_481 AEKS8_481.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint012.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_482 AEKS8_482.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:14:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Joint012.Reference1 missing element reference RackCase___Kopie__2_#AEKS8_482 AEKS8_482.;Edge4;:H-f38,E;:Hf38:9,E.Edge4
12:16:19  <PropertyLinks> PropertyLinks.cpp(494): RackCase___Kopie__2_#Extend001.baseObject missing element reference RackCase___Kopie__2_#Extend ;#51dd:15;:Hf0c,F;:H,F.Face38
[... cut ...]
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * DynamicData 2.74.0
  * fasteners 0.5.41
  * kicadStepUpMod 11.7.3
  * sheetmetal 0.7.24
`

tobiasfalk, 2025-08-24:
And it breakes even more dramatic in the stable release:

[image]

```
12:20:40  <PropertyLinks> PropertyLinks.cpp(449): RackCase#Joint021.Reference2 missing element reference RackCase#Top Top.;#129c0:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge52
12:20:40  <PropertyLinks> PropertyLinks.cpp(449): RackCase#Joint021.Reference2 missing element reference RackCase#Top Top.;#129c0:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge52
12:20:40  <PropertyLinks> PropertyLinks.cpp(429): RackCase#Joint022.Reference2 auto change element reference RackCase#Top Top.;#129bf:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge51 -> Top.;#13519:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge52
12:20:40  <PropertyLinks> PropertyLinks.cpp(429): RackCase#Joint022.Reference2 auto change element reference RackCase#Top Top.;#129bf:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge51 -> Top.;#13519:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge52
12:20:40  <PropertyLinks> PropertyLinks.cpp(449): RackCase#Joint023.Reference2 missing element reference RackCase#Top Top.;#129be:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge50
12:20:40  <PropertyLinks> PropertyLinks.cpp(449): RackCase#Joint023.Reference2 missing element reference RackCase#Top Top.;#129be:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge50
12:20:55  <PropertyLinks> PropertyLinks.cpp(429): RackCase#Joint022.Reference2 auto change element reference RackCase#Top Top.;#13519:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge52 -> Top.;#13c8b:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge51
12:20:55  <PropertyLinks> PropertyLinks.cpp(429): RackCase#Joint022.Reference2 auto change element reference RackCase#Top Top.;#13519:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge52 -> Top.;#13c8b:b;:Hf4e,E;:H,E;:Hf50:5,E.Edge51
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Installed mods: 
  * DynamicData 2.74.0
  * fasteners 0.5.41
  * kicadStepUpMod 11.7.3
  * sheetmetal 0.7.24

```

MisterMakerNL, 2025-08-25:
This should be a blocker @maxwxyz if this isn't fixed, I tried using Assembly the past weeks and got nothing but very frustrated. So I don't think we can release it as is.

maxwxyz, 2025-08-25:
by definition this cannot be a blocker if we released 1.0 with this. There is still time for a fix PR after feature freeze though.

drwho495, 2025-08-25:
> This should be a blocker @maxwxyz if this isn't fixed, I tried using Assembly the past weeks and got nothing but very frustrated. So I don't think we can release it as is.

What design workbench are you using, what operations are you using and are any of those added with addons?

In the latest weekly builds all of the failures are caused by a dumb regression I made, or because of partdesign operations that are missing element map generation (specifically pipes and revolutions)

drwho495, 2025-08-26:
> Out of interest, is there a need from Addon developers to do something to fix these issues if they happen in combination with their WB?

For some general things, yes they could, but not for everything that is needed for stuff like the sheet metal addon

Most of the edge cases can be fixed in the freecad code base, which is what I am planning to do.

MisterMakerNL, 2025-08-26:
I did a simple test and any simple change to the model will break all constraints to any geometry. It doesn't even have to be linked too the feature that creates the face the constraint is linked too.
I'm just using part design.
If I need to test a pr please let me know!

drwho495, 2025-08-26:
> I did a simple test and any simple change to the model will break all constraints to any geometry. It doesn't even have to be linked too the feature that creates the face the constraint is linked too.
> I'm just using part design.
> If I need to test a pr please let me know!

That's not good, could you send files for these simple tests?

maxwxyz, 2026-07-18:
@drwho495 is this fixed by v2?