FreeCAD/FreeCAD #23009: Sketch circle shortcut missing

G0rocks, 2025-08-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you hover over the circle command the shortcut is not displayed.
Would be cool if somebody could add this since I sometimes forget these shortcuts and the hover to find them function is really helpful for that.

[image]

Though the shortcut G,C works.

Note how the shortcut is displayed when hovering over the line command:

[image]

Note also that the shortcuts are displayed if you press the dropdown button:

[image]

This applies to the other methods of making a circle/ellipsoid as well:

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.7
  * Manipulator 1.5.7
  * Render 2024.1.23
```

### Subproject(s) affected?

None

### Anything else?

The only potentially similar issue I foudn was this one:
https://github.com/freecad/freecad/issues/22282

But I feel like that is different

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-08-10:
Related/duplicate: #22610.

maxwxyz, 2025-08-12:
I think it is not showing this currently as this is a command group, not the single command. Maybe command groups in general could show the shortcuts for the currently displayed command.

G0rocks, 2025-08-12:
That would be cool. I don't know how the implementation is though so potentially it is easier to just add the shortcut for these specific ones and then for other command groups add them also when we find them?
Though a generalized solution is more elegant and future-proof probably.

G0rocks, 2025-08-12:
Ooops, I pressed close instead of comment, that was an accident.
I also do not have permission to reopen this issue. Sorry about this. Potentially you could reopen this issue?

matt-taylor-git, 2025-09-19:
Took a shot at this with: https://github.com/FreeCAD/FreeCAD/pull/24043

furgo16, 2025-09-20:
@matt-taylor-git this works great, thank you!

Would you fancy taking a shot at fixing the same issue for Python workbenches as well? https://github.com/FreeCAD/FreeCAD/issues/22610#issuecomment-3314637581