FreeCAD/FreeCAD #23013: Dropdown menu in any workbenches is not visible

p0lp44r, 2025-08-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Tested formally in the Sketcher workbench but appears also in the Part Design.

If you want use any drop down menu in the toolbar section, the drop down will pop up at first time but never if you click on it twice or more.

I cloud reset the problem by deleting all config files in ~/.config/Freecad but the problem come back again. Restart doesn't work at all.

No errors in Freecad console are shown. Next try is to start Freecad via bash to see if there are any hints.

Please check the screen video for exact details.

https://github.com/user-attachments/assets/344f3b6d-4fbf-4faf-b04e-39ed2da43ec2

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.2.39319 (Git)
Build type: Release
Branch: makepkg
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, OCC 7.9.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * 3D_Printing_Tools
  * Curves 0.6.68
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ogruendel, 2025-08-10:
Can confirm I'm also affected, however for me not even the first dropdown menu shows. All items seem to still be clickable but are not rendered.

```
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.2.39319 (Git)
Build type: Release
Branch: makepkg
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, OCC 7.9.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

ogruendel, 2025-08-11:
I seem to have found a workaround, which is to launch FreeCAD in X11 mode, instead of wayland:
```
QT_QPA_PLATFORM=xcb freecad
```

p0lp44r, 2025-08-11:
I will test it tomorrow as well. Is this in the freecad config?

closed 2025-08-12 by maxwxyz