FreeCAD/FreeCAD #23018: Bug in Draft Shape String with FreeCAD language set to Spanish (not Spain) – Fonts do not appear

exezol, 2025-08-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I set FreeCAD’s language to Spanish (Latin America, not Spain), the Draft Shape String tool cannot locate any fonts. In the file browser, only folders are shown — no font files appear — regardless of the font type or format.

When I switch the language to Spanish (Spain), the fonts appear and the tool works as expected.

Steps to Reproduce:

Set FreeCAD language to Spanish (not Spain).

Open Draft → Shape String.

Try to select a font.

Observe that no font files are displayed, only folders.

Expected Behavior:
Font files should be visible and selectable, just as when using Spanish (Spain) or other languages.

Actual Behavior:
No font files are displayed; only folders are shown.

### Full version info

```shell
OS: Windows 11 build 22000
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * kicadStepUpMod 11.7.3
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-11:
@exezol could you please try this in the latest dev version: https://github.com/FreeCAD/FreeCAD/releases/

maxwxyz, 2025-08-11:
Thanks, closing then as fixed in 1.1. The error message is https://github.com/FreeCAD/FreeCAD/issues/22777