FreeCAD/FreeCAD #23025: Assembly : Changing joint type in property panel is not reflected in model pane.

BanceFC, 2025-08-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It is possible to change the joint type in the property panel, but this is not reflected in the model pane. 
The label remains as the original. 
The Base label does not reflect the Type change. 
The joint type does change but not the label, this is confusing.

[image]

[Model .TXT](https://github.com/user-attachments/files/21710328/Model.TXT)

File is FCStd renamed to TXT.

### Full version info

```shell
OS: Manjaro Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.42837 (Git) AppImage
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1
Installed mods: 
  * fasteners.backup1738584191.325799 0.5.36 (Disabled)
  * Curves 0.6.64
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * Vars 0.0.2.beta2
  * dodo 1.0.1
  * A2plus 0.4.68
  * sheetmetal 0.7.24
  * Assembly4 0.60.1
  * boltsfc 2022.11.5
  * fcgear 1.0.0
  * AirPlaneDesign 0.4.1
  * frame 0.1.1
  * CurvedShapes 1.0.13
  * DynamicData 2.74.0
  * kicadStepUpMod 11.7.2
  * Plot 2024.11.26
  * AddonManager 2025.7.17
  * Nodes 0.1.36
  * BOLTSFC 2022.11.5
  * Silk 0.2.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

This makes searching for joints a little difficult!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-12:
>  But is not reflected if previously changed.

@BanceFC would you mind updating ticket title ITC ? TIA

BanceFC, 2025-08-12:
Title changed

BanceFC, 2025-08-12:
This is a bug not a feature request. If the joint type changes then the label should change to reflect that.

maxwxyz, 2025-08-12:
That can only happen, if the label was not changed by the user

BanceFC, 2025-08-12:
"That can only happen, if the label was not changed by the user"

I did not change the label only the joint type!

As it stands, the only indication of the joint type available in the model pane is the label. 

If the joint type is changed then the label should change to reflect that. Or some other method of identifying the joint should exist.

A user looking to edit a slider joint will search for slider in the model pane, if the label says revolute what chance does the user have of finding the joint? 

The label does change if the dialogue is used, provided that it has not been previously changed in the property panel. This is inconsistent at least, moreover confusing. 

The Tree structure in assembly could do with some work generally, identifying joints and their associated elements is difficult at the moment, but that would be better in a separate feature request.

BanceFC, 2025-08-12:
As I said that (joint task box) only works if the joint hasn't been changed in the property panel, everywhere else in FC, properties are updated and the changes reflected if the property panel is used.
If it is not a bug OK, but we should look for consistency.

closed 2025-08-25 by chennes