FreeCAD/FreeCAD #23044: 1.1dev (BIM example) onDocumentRestored has no Placement attribute

andreymrt, 2025-08-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Running weekly dev build, starting the BIMExample.FCStd raises the following python error:

```
00:46:58  pyException: Traceback (most recent call last):
  File "C:\cad\FreeCAD_weekly-2025.08.07-Windows-x86_64-py311\Mod\BIM\ArchWindow.py", line 193, in onDocumentRestored
    self.basePos = obj.Base.Placement.Base
                   ^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'Placement'
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
```

### Subproject(s) affected?

BIM

### Anything else?

This is using the built-in BIMExmample.FCStd sample

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-08-11:
I can confirm the error too.

I am not completely sure, but I don't think the error was there in the dev version ca. 6 months ago. But i might be mistaken, as the error is hidden in between warnings.

paullee0, 2025-08-16:
Where to find the model file in concern?  Thanks.

paullee0, 2025-08-16:
This one?  Nothing wrong here.

FreeCAD_weekly-builds-42721-conda-Linux-x86_64-py311.appimage on Fedora 40

[image]

semhustej, 2025-08-16:
The error is not displayed in the notification widget. It is visible only in Report View window.

paullee0, 2025-08-16:
Nothing in my Report View for Git 42721though.

semhustej, 2025-08-16:
Maybe a windows-only issue then? I am able to reproduce the issue in Safe Mode too.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

paullee0, 2025-08-16:
OK, it happens I have the fix on my system which has been submitted for PR, so no error whichever the build are :
-https://github.com/FreeCAD/FreeCAD/pull/23014  ([ArchWindow] Clone and Link to Support Sill Change)

There is an earlier feature added about Sill, a series of regression was created.

The above is the latest fix which is pending reviewed and merged - this somehow prove the fix works :)

(Removed the 'fix' on my system, now the error is shown below)

[image]

PR #23014 (2025-08-25): [ArchWindow]  Clone and Link to Support Sill Change
Problems / Course of Developement:

  1. Allow changing Sill feature was added in
  - https://github.com/FreeCAD/FreeCAD/pull/21005

  2. Bug fix was added in
  - https://github.com/FreeCAD/FreeCAD/commit/f6bba5d58ec7788c37977b97d7db40ef1e1ff499

  3. Further Problems were identified as discused in
  - https://github.com/FreeCAD/FreeCAD/pull/21005#issuecomment-3010959162

  'New Observation-

      1. Clone or Link to a Window does not work and return error
      2. This PR/Commit (21005) translate the Base Sketch when Sill property is change [... cut]