FreeCAD/FreeCAD #23049: Revit camera controls no longer make the rotation pivot the zoom pivot.

Lunarien, 2025-08-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In FreeCAD 1.0, when i used the Revit camera controls, rotating around a body would show a pivot around which the rotation is happening. That action of rotating around it would make the zoom center be in the same place as the pivot.

In FC 1.0.1, you now have to **middle click** wherever you want the zoom center/zoom pivot to be. This is annoying and makes the design process slower.
If Revit had a change in its camera navigation, I don't see why FreeCAD would copy that, since it just makes things worse.

Have a look at a recording i made. It shows a big problem with this:

https://github.com/user-attachments/assets/4f55b75c-2354-4c71-ad83-689359586d08

I don't wanna have to do an additional middle click every time I need to zoom to a different area. On top of that, middle clicks are kinda hard to perform on some computer mice.

This probably affects **all workbenches** with a 3D viewport

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I installed FreeCAD 1.0 to show what it was like:

https://github.com/user-attachments/assets/6481e7b0-5a19-455f-9d3b-6fd9a8937e3f

(it was nice)

I also just tested Blender controls and found its not just a Revit controls thing, it happens with Blender controls as well. I am guessing it **happens with every camera controls setting**. I don't wanna edit the text i wrote. It makes sense for any control scheme, really.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Lunarien, 2025-08-12:
yes, here:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.33
  * ThreadProfile 1.92.0

Rexbas, 2025-08-12:
Regression introduced in #18992. It is not specific to Revit style.

closed 2025-08-25 by chennes

Lunarien, 2025-08-29:
I can confirm it works as expected in the latest dev version. Thanks for this :+1:.