FreeCAD/FreeCAD #23050: Crash when clicking on Sketch's "Attachment Support"

MihaMarkic, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There are two issues with attached file.
1. Sketch002 started reporting an error for no apparent reason (to me). It works seemingly.
2. FreeCAD would crash when selecting Sketch002 and clicking on [...] in Attachment Support attribute value.

[bug.zip](https://github.com/user-attachments/files/21728241/bug.zip)

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MihaMarkic, 2025-08-12:
Hm, it seems that FreeCAD crashes on all [...] buttons in properties. Fedora 42 / KDE Plasma here.

xtemp09, 2025-08-12:
No crash here.

It is possible that this problem is flatpak-specific.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git) AppImage
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Trackball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.8773/1
```

maxwxyz, 2025-08-12:
does it also crash in the weekly https://github.com/FreeCAD/FreeCAD/releases ?

ErtugrulEge, 2025-08-12:
When I open in latest weekly skecth shows error messages but no crash

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61

[image]

maxwxyz, 2025-08-12:
@PaddleStroke could this be because of the change to core datums?

MihaMarkic, 2025-08-12:
> does it also crash in the weekly https://github.com/FreeCAD/FreeCAD/releases ?

If I run it as AppImage it does not crash. I have give 1.0.2 AppImage a try to see whether it's version or flatpak

ErtugrulEge, 2025-08-12:
> > When I open in latest weekly skecth shows error messages but no crash
> 
> Click on [...] button in Sketch's [...] button in Attachment Support attribute value. Then it crashes. I initially assumed it was only because errors in sketch but now I see that it crashes on all [...] button clicks for some reason.

if I didn't misunderstand I couldn't reproduce the crash

https://github.com/user-attachments/assets/9186f233-a3aa-4411-81ee-c60c0095a0c5

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

kadet1090, 2025-08-12:
Thanks, I'll check what's going

kadet1090, 2025-08-12:
@xtemp09 that is not the same issue. The one that you mention is due to moving Pad out of body - which should not be possible. Should be fixed too, but not related.

IGLOU-EU, 2025-08-22:
Hi there,
Same for me, with my sketch004 I need to change the attachment support, but this segfault.
Thans for FreeCAD and for your time :)

```
Sketch004: AttachEngine3D: subshape not found Pocket.?Face7
Sketch004: AttachEngine3D: subshape not found Pocket.?Face7
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x41140) [0x7f5124242140]
#1  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x4a7f62) [0x7f5125ea7f62]
#2  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x4a92f5) [0x7f5125ea92f5]
#3  /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6(+0x681089) [0x7f5125681089]
#4  0x7f5125f5c004 in QTreeView::selectionChanged(QItemSelection const&, QItemSelection const&) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x154
#5  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x20dfee) [0x7f5124a0dfee]
#6  0x7f5124c581fd in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x3d
#7  0x7f5124c649c7 in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x287
#8  0x7f5124c63ff2 in QItemSelectionModel::select(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x62
#9  0x7f5125f6277f in QTreeWidgetItem::setSelected(bool) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x9f
#10  0x7f5127e818c5 in Gui::Dialog::DlgPropertyLink::onSelectionChanged(Gui::SelectionChanges const&) from /app/freecad/lib/libFreeCADGui.so+0x195
#11  0x7f5127e85324 in Gui::Dialog::DlgPropertyLink::init(App::DocumentObjectT const&, bool) from /app/freecad/lib/libFreeCADGui.so+0xb34
#12  0x7f51280feeec in Gui::PropertyEditor::LinkLabel::onEditClicked() from /app/freecad/lib/libFreeCADGui.so+0x9c
#13  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x20dfee) [0x7f5124a0dfee]
#14  0x7f5125ce8434 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x44
#15  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x2e86bf) [0x7f5125ce86bf]
#16  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x2e9dcf) [0x7f5125ce9dcf]
#17  0x7f5125ce9fe5 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x115
#18  0x7f5125c09a08 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x2b8
#19  0x7f5125ba167d in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x8d
#20  0x7f5125bac09b in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0xe9b
#21  0x7f5127d54040 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xd0
#22  0x7f51249a5808 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x168
#23  0x7f5125baa64c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x1ec
#24  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x21e74f) [0x7f5125c1e74f]
#25  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x2211c8) [0x7f5125c211c8]
#26  0x7f5125ba167d in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x8d
#27  0x7f5127d54040 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xd0
#28  0x7f51249a5808 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x168
#29  0x7f5125202bd2 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x3d2
#30  0x7f512526d29c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0xac
#31  /usr/lib/x86_64-linux-gnu/libQt6XcbQpa.so.6(+0x64046) [0x7f511f049046]
#32  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x633f6) [0x7f51233133f6]
#33  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xc40c7) [0x7f51233740c7]
#34  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f5123312883]
#35  0x7f5124cc29b3 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x73
#36  0x7f51249b31e3 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x1a3
#37  0x7f51249ae7c1 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0xb1
#38  0x7f5127c697bd in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0xd4d
#39  FreeCAD(+0x96e3) [0x55cb109336e3]
#40  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2a188) [0x7f512422b188]
#41  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7f512422b24b]
#42  FreeCAD(+0x9b15) [0x55cb10933b15]
```

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * Marz 0.1.16
  * FreeGrid 2.2.0
  * QuickMeasure 2022.10.28
  * freecad.gears 1.3.0
  * STEMFIE 0.3.1
  * 3DfindIT 1.2.0
```