FreeCAD/FreeCAD #23064: Cannot apply parallel constraint on flat face with two angles in Assembly workbench

MisterMakerNL, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the Assembly workbench, it is not possible to apply a parallel constraint to a flat face if the face has two angles, even though the face itself is geometrically flat. This limits assembly workflows where complex flat faces are involved. This restriction makes it difficult or impossible to align such parts using parallel constraints, reducing the flexibility for more advanced or non-rectangular geometries.

https://github.com/user-attachments/assets/4333dba4-28a6-4095-af62-dd324c06107f

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42917 +26 (Git)
Build date: 2025/08/12 11:53:17
Build type: Release
Branch: Stylesheet_fixes2
Hash: 70b077311629e39eaf1774b2772c8b894a0428fc
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD_Stylesheet_roaming.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/69.8349/2
Installed mods: 
  * AddonManager 2025.8.8
  * Assembly3 0.12.2
  * BIM021 2021.12.0
  * Dracula 0.0.6
  * fasteners 0.5.41
  * FeedsAndSpeeds 0.6.0
  * freecad_metal_workbench 0.0.1
  * Gridfinity 0.11.9
  * Iconpack 2023.10.25
  * IconThemes (Disabled)
  * lattice2 1.0.0
  * OpenTheme 2025.5.20
  * Render 2024.12.15
  * scanObjects
  * sheetmetal 0.4.21
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2
  * woodworking 1.1.0.dev.1.1.1.2025.08.11
```

### Subproject(s) affected?

Assembly

### Anything else?

[test parallel.zip](https://github.com/user-attachments/files/21742783/test.parallel.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-08-12:
@PaddleStroke

closed 2025-08-13 by maxwxyz

drwho495, 2025-08-13:
if it is truly not a planar face, then you can collect the edges of the face and create another plane with the face class, or you could create a plane from the vertexes of some of edges.

pieterhijma, 2025-09-09:
> Try doing the angled faces with the draft command. The tapering is producing faces like a loft, which are not planar IIRC. [@pieterhijma](https://github.com/pieterhijma) is this an OCC way of doing things and could FreeCAD have a workaround?

Based on the code below, the faces do not seem to be planar. Otherwise, the type id of the surface would be `'Part::GeomPlane'` instead of `'Part::GeomBSplineSurface'`. My guess is that the Assembly workbench does not handle this case. It could do, for example:

``` python
doc = App.ActiveDocument
face1 = doc.Body.getSubObject("Face2")
face1.Surface.TypeId
# 'Part::GeomBSplineSurface'
pointFace1 = face1.CenterOfMass # (or some other coordinate)
u, v = face1.Surface.parameter(pointFace1)
normalVectorFace1 = face1.normalAt(u, v)
normalVectorFace1
# Vector (-0.8657465077463428, 0.24192189559966756, -0.4381287262374784)
face2 = doc.Body001.getSubObject("Face3")
face2.Surface.TypeId
# 'Part::GeomBSplineSurface'
pointFace2 = face2.CenterOfMass
u, v = face2.Surface.parameter(pointFace2)
normalVectorFace2 = face2.normalAt(u, v)
normalVectorFace2
# Vector (0.0, 0.24192189559966756, -0.9702957262759966)
```

And align based on this.

By the way, this has been closed as a duplicate, but I tested this with

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.43406 (Git)
Build date: 2025/09/09 06:43:51
Build type: Unknown
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
```

Which contains the fix of #21242 and the parallel constraint doesn't work.