FreeCAD/FreeCAD #23065: Some faces disappear before using fillet tool!

tikooriginal-pixel, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After applying fillet on a part, some faces disappeared like the reconstruction had failed!

[image]
[image]

### Full version info

```shell
OS: macOS 15.6
Architecture: arm64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.60n
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-08-14:
@tikooriginal-pixel Could you provide the file you show in your screenshot?

In the meantime I have [made a crude recreation of it](https://github.com/user-attachments/files/21774735/fillet_iz_broke.FCStd.zip) and can https://github.com/FreeCAD/FreeCAD/labels/Status%3A%20Confirmed (with `1.1.0dev.42885` (2025/08/08 15:11:51)) the general brokenness of geometry. https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC stuff as usual.

<details>
<summary>Full version info</summary>

```
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git) AppImage
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.38
  * FreeCAD-Ribbon 1.9.0.1
  * SearchBar 1.0.2
```

</details>

Though your case is most likely a duplicate with any one of these, in fact they may all be the same bug altogether:
* #11082
* #19383
* #19703
* #17845
* #17682
* #17261

tikooriginal-pixel, 2025-08-15:
En effet, il semblerait que d'autres autos aient des problèmes liés au congés! J'ai essayé la dernière version DEV et il semblerait qu'elle apporte certaines améliorations! Si je fait mes fillets 1 par 1, la reconstruction de la pièce est un succès!
Je partage le fichier avec le bug, si jamais ce peut aider :)

[Archive.zip](https://github.com/user-attachments/files/21804332/Archive.zip)

maxwxyz, 2025-08-16:
@pieterhijma FYI OCC