FreeCAD/FreeCAD #23066: PartDesign: Up to face & Up to shape modes are too similar to be different

pierreporte, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is about `PartDesign_Pad`, `PartDesign_Pocket`, `PartDesign_Revolution`, and `PartDesign_Groove`.

Both modes are orthogonal in their feature set and both are incomplete, but they are extremely similar in nature. Up to face allows to select just one face or plane, while up to shape allows to select multiple faces from a given solid. Up to face can have an offset any offset while up to shape can’t. These modes should be merged into one mode, up to geometry, that does everything:

* selecting multiple geometries to limit the extrusion, not just one face;
* extending the type of geometry that can be selected, like points, lines, parts of sketches, etc., but also solids;
* selection solids would replace the “select all faces” modes of current up to shape, while selecting individual faces removes the currently mandatory solid selection;
* offset option like up to face.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * Curves 0.6.63
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * FusedFilamentDesign 0.25.220
```

### Subproject(s) affected?

PartDesign

### Anything else?

Discussed with @FreeCAD/design-working-group 

Supersedes #5724 and #16176

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-08-12:
Ping @maxwxyz

pierreporte, 2025-08-29:
@PaddleStroke what do you think about this, since your PR is now merged?

PaddleStroke, 2025-08-29:
I agree. But my plate is kind of full right now :)

FlachyJoe, 2025-08-29:
> offset option like up to face.
Needs a 3D offset. Maybe more efficient to implement a 3D offset on SubShapeBinder and direct the user's working flow to Binder with Offset and (new) UpToSolid. I'm not comfortable with a unique complex feature.

maxwxyz, 2026-11-07:
UnKabaraQuiDev added a two-face curved-surface example in #33547 on Ubuntu 25.10, 27.1.0dev.49206 +1: leave 1 mm of material with a Pocket. A SubShapeBinder lets Up to shape use both faces, but has no offset option; Up to face has the offset but only accepts one face. They suggest either adding the offset to Up to shape or allowing multiple faces in Up to face.