FreeCAD/FreeCAD #23068: Both simulations fail if toolbit's cutting edge length == the bit length

freckletonj, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

For any tool bit where you've defined the cut height the same as the bit length:

1. The old simulator gives a handy error that something's up, and by experimentation I determined it was when cut height==bit len

2. The new simulator silently replaces the actual bit with a 1/4" bit (or the last working bit?), and simulates it chewing up your part

3. This also shows up when you try to visualize a bit from the worktree. A cut=len bit shows as a 1/4" endmill for me, but once you fix it it properly displays the right geometry. There are other issues with bit display on AppImage, where it doesn't look for geometry files in the right place, so I had to copy them into `~/.local/share/FreeCAD/...` myself, from the git repo.

I typically have been setting cut height == bit len because I don't think FC takes the bit len into account, and I don't think it calculates collision detection, or reaching too far into a part, etc. Also, I won't reliably chuck the bit at the same location, so this info is kind of useless (i think?)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.68
```

### Subproject(s) affected?

CAM

### Anything else?

If collision detection were to exist (https://github.com/FreeCAD/FreeCAD/issues/22178 amazing!) this would need to be fixed

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaxTux, 2025-08-14:
I cannot reproduce on
```
OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git)
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Free Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * fasteners 0.5.38
```

Please try the latest weekly build, the bug might be already fixed.

maxwxyz, 2025-08-16:
Please test in the latest dev version: https://github.com/FreeCAD/FreeCAD/releases/ and share an example file

freckletonj, 2025-08-16:
I can't checkout the latest right now, so this is `1.0.1`, so if this was fixed great!

In the worktree, this 1/32" bit shows as a 1/4", even though it's the only bit in the job:

[image]

The simulations fail to respect tool geometry:

[image]

[image]

If I edit the tool shape from the work tree, you can see it shows a 1/4" bit if the cutting edge at tool length are the same:

[image]

But it updates once the cutting edge length doesn't equal the total length:

[image]

[t13_fail_toolshapes.zip](https://github.com/user-attachments/files/21814111/t13_fail_toolshapes.zip)

freckletonj, 2025-08-16:
Do the inputs not make sense? IIUC no algorithm takes bit length into account. And at least on hobbyist machines, because you chuck the bit differently each time, they can't really make sense, so I've just defaulted to making my bits have 0 length above the cutting edge, but I see how that's problematic now.

If the bug is that I had a "non-sensical" tool definition, we should probably get a better error when defining the tool, and not allow the tool to be made since it will break things downstream (just my 2 cents)

PaxTux, 2025-08-17:
> Do the inputs not make sense?

Cutting Edge Height < Total length -> makes sense
Cutting Edge Height > Total length -> doesn't makes sense
Cutting Edge Height = Total length -> it's debatable. Personally I think it would be ok if FreeCAD accepted this edge case. I also think it's ok if it's not accepted, as long as it prevent user to write such parameters or gives a clear feedback what's wrong (currently that's not the case).

> ...we should probably get a better error when defining the tool, and not allow the tool to be made since it will break things downstream (just my 2 cents)

I agree.

sliptonic, 2025-10-16:
@shaise can you comment?

freckletonj, 2026-10-25:
still relevant, and simply requires some logic to error early if the tool length is LTE cutting edge height to ensure the shank length is >0.

luzpaz, 2026-10-25:
Can someone kindly provide a simple example file to attach to this ticket?  

Edit: Status of this ticket: see shaise's diagnosis in https://github.com/FreeCAD/FreeCAD/issues/23068#issuecomment-3462256588

Connor9220, 2026-10-25:
How on earth is the over all length of the tool bit the cutting edge height?  Your describing impossible geometry.  This isn't a bug in the sims.  This is a issue with use input.

freckletonj, 2026-10-26:
Exactly, current behavior gives incorrect fallbacks and inexplicable error messages, and so impossible geometries should fail sooner in the chain and give appropriate error messages.