FreeCAD/FreeCAD #23069: FreeCAD 1.1 TechDraw Looks for Template Files in FreeCad 0.2 Directory

akotowski, 2025-08-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When trying to create a TechDraw Page FreeCAD 1.1 tries to access C:/Program Files/FreeCAD 0.20/data/Mod/TechDraw/LineGroup/LineGroup.csv, which does not exist.  I did have 0.2 installed but have uninstalled it. I also the get BOPAlgo_AlertSelfInterferingShape.

The object I am drawing is a 1" x 2" x 3" cube.

[Test.zip](https://github.com/user-attachments/files/21744647/Test.zip)

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.134/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-08-13:
Specifically with regard to the Templates folder issue, when you de-installed 0.20.x there was a tickbox to remove all user data which wasn't ticked so the configs from that version were kept to be picked up by any future install of FreeCAD so that all worked by design. In order to correct the location of the TechDraw Templates, first ensure the TechDraw workbench has been loaded and then from the `Edit` menu select `Preferences` and change the _Template Directory_ as per screenshot (I use one folder for all versions of FreeCAD hence why it's a custom folder):

[image]

For more detailed info see https://wiki.freecad.org/TechDraw_Preferences

closed 2025-08-13 by maxwxyz

akotowski, 2025-08-13:
> another case for version depending config and mods @kadet1090

Thanks.

Sorry.
I must have not had TechDraw loaded when I  edited preferences.

Andy Kotowski