FreeCAD/FreeCAD #23075: Preferences: Crash when clicking "OK" or "Apply" buttons

ryankembrey, 2025-08-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:
1) Open the Preferences dialog
2) Click either the "OK" or "Apply" button

### Full version info

```shell
OS: Fedora Linux 41 (Sway) (sway/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.42916 (Git)
Build date: 2025/08/11 19:38:58
Build type: Release
Branch: main
Hash: 44a8ec2b6559fa5195c8573b4bd749582e4f29a0
Python 3.13.5, Qt 6.8.3, Coin 4.0.2, Vtk 9.2.6, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.8.0
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * OpenTheme 2025.5.20
  * CfdOF 1.31.4
  * sheetmetal 0.7.24
  * fasteners 0.5.39
  * Render 2024.12.15
  * Telemetry 1.0.2
  * DFM
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ryankembrey, 2025-08-13:
@ErtugrulEge FreeCAD crashes

maxwxyz, 2025-08-13:
I cannot reproduce, can you check on latest weekly in safe mode?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * AddonManager 2025.8.5
  * Cables 0.2.1
  * CfdOF 1.31.5
  * CurvedShapes 1.0.13
  * Curves 0.6.68
  * Defeaturing 1.2.6
  * dodo 1.0.1
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * Gridfinity 0.11.9
  * Manipulator 1.6.0
  * OpticsWorkbench 1.2.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.24
  * Silk 0.2.0
  * Telemetry 1.0.2
```

luzpaz, 2025-08-13:
@ryankembrey can you start from the CLI and report what the output shows when crash is re-triggered ?

ryankembrey, 2025-08-13:
Output in CLI when clicking "Apply"
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x1a490) [0x7fbadf226490]
#1  0x7fbae3930105 in Gui::Document::getViewProvidersOfType(Base::Type const&) const from /usr/local/lib64/libFreeCADGui.so+0x15
#2  0x7fba3e637448 in PartGui::DlgSettings3DViewPart::saveSettings() from /usr/local/lib64/PartGui.so+0x98
#3  0x7fbae3bd2a41 in Gui::Dialog::DlgPreferencesImp::applyChanges() from /usr/local/lib64/libFreeCADGui.so+0x141
#4  /lib64/libQt6Core.so.6(+0x15ad2e) [0x7fbadf95ad2e]
#5  0x7fbae0c59741 in QDialogButtonBox::clicked(QAbstractButton*) from /lib64/libQt6Widgets.so.6+0x41
#6  /lib64/libQt6Widgets.so.6(+0x25a216) [0x7fbae0c5a216]
#7  /lib64/libQt6Core.so.6(+0x15ad2e) [0x7fbadf95ad2e]
#8  0x7fbae0b82794 in QAbstractButton::clicked(bool) from /lib64/libQt6Widgets.so.6+0x44
#9  /lib64/libQt6Widgets.so.6(+0x182a1f) [0x7fbae0b82a1f]
#10  /lib64/libQt6Widgets.so.6(+0x18410f) [0x7fbae0b8410f]
#11  0x7fbae0b84325 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib64/libQt6Widgets.so.6+0x115
#12  0x7fbae0aa4698 in QWidget::event(QEvent*) from /lib64/libQt6Widgets.so.6+0x2b8
#13  0x7fbae0a3d9b8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x88
#14  0x7fbae0a48b3b in QApplication::notify(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0xebb
#15  0x7fbae39a0030 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xa0
#16  0x7fbadf8f3f30 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x150
#17  0x7fbae0a470cc in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib64/libQt6Widgets.so.6+0x1ec
#18  /lib64/libQt6Widgets.so.6(+0xb908c) [0x7fbae0ab908c]
#19  /lib64/libQt6Widgets.so.6(+0xbbc10) [0x7fbae0abbc10]
#20  0x7fbae0a3d9b8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x88
#21  0x7fbae39a0030 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xa0
#22  0x7fbadf8f3f30 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x150
#23  0x7fbae00eb2e3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib64/libQt6Gui.so.6+0x3e3
#24  0x7fbae015383c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Gui.so.6+0xac
#25  /lib64/libQt6Gui.so.6(+0x6cd714) [0x7fbae06cd714]
#26  /lib64/libglib-2.0.so.0(+0x4028c) [0x7fbadbef128c]
#27  /lib64/libglib-2.0.so.0(+0xa08b8) [0x7fbadbf518b8]
#28  /lib64/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fbadbef2783]
#29  0x7fbadfbf2c33 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Core.so.6+0x73
#30  0x7fbadf901313 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Core.so.6+0x1b3
#31  0x7fbae0cdd1ed in QDialog::exec() from /lib64/libQt6Widgets.so.6+0x1fd
#32  0x7fbae3a4e06e in StdCmdDlgPreferences::activated(int) from /usr/local/lib64/libFreeCADGui.so+0x7e
#33  0x7fbae3a18cd3 in Gui::Command::_invoke(int, bool) from /usr/local/lib64/libFreeCADGui.so+0x1f3
#34  0x7fbae3a19106 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/local/lib64/libFreeCADGui.so+0x126
#35  /lib64/libQt6Core.so.6(+0x15ad2e) [0x7fbadf95ad2e]
#36  0x7fbae0584794 in QAction::triggered(bool) from /lib64/libQt6Gui.so.6+0x44
#37  0x7fbae05879af in QAction::activate(QAction::ActionEvent) from /lib64/libQt6Gui.so.6+0x10f
#38  /lib64/libQt6Widgets.so.6(+0x23b414) [0x7fbae0c3b414]
#39  /lib64/libQt6Widgets.so.6(+0x244450) [0x7fbae0c44450]
#40  0x7fbae0aa4698 in QWidget::event(QEvent*) from /lib64/libQt6Widgets.so.6+0x2b8
#41  0x7fbae0a3d9b8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x88
#42  0x7fbae0a48b3b in QApplication::notify(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0xebb
#43  0x7fbae39a0030 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xa0
#44  0x7fbadf8f3f30 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x150
#45  /lib64/libQt6Widgets.so.6(+0x245301) [0x7fbae0c45301]
#46  0x7fbae0c456fe in QMenu::mouseReleaseEvent(QMouseEvent*) from /lib64/libQt6Widgets.so.6+0x5e
#47  0x7fbae0aa4698 in QWidget::event(QEvent*) from /lib64/libQt6Widgets.so.6+0x2b8
#48  0x7fbae0a3d9b8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x88
#49  0x7fbae0a48b3b in QApplication::notify(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0xebb
#50  0x7fbae39a0030 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xa0
#51  0x7fbadf8f3f30 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt6Core.so.6+0x150
#52  0x7fbae0a470cc in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib64/libQt6Widgets.so.6+0x1ec
#53  /lib64/libQt6Widgets.so.6(+0xb964b) [0x7fbae0ab964b]
#54  /lib64/libQt6Widgets.so.6(+0xbbc10) [0x7fbae0abbc10]
#55  0x7fbae0a3d9b8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt6Widgets.so.6+0x88
#56  0x7fbae39a0030 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib64/libFreeCADGui.so+0xa0
[... cut ...]
#65  0x7fbadfbf2c33 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Core.so.6+0x73
#66  0x7fbadf901313 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt6Core.so.6+0x1b3
#67  0x7fbadf8fc90e in QCoreApplication::exec() from /lib64/libQt6Core.so.6+0xae
#68  0x7fbae38be503 in Gui::Application::runApplication() from /usr/local/lib64/libFreeCADGui.so+0x973
#69  FreeCAD() [0x402e77]
#70  /lib64/libc.so.6(+0x3488) [0x7fbadf20f488]
#71  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7fbadf20f54b]
#72  FreeCAD() [0x403135]
```

maxwxyz, 2025-08-13:
Seems to be the same reason than https://github.com/FreeCAD/FreeCAD/issues/23015
Can you load CAM WB before opening preferences and then try again?

closed 2025-08-13 by maxwxyz

ryankembrey, 2025-08-13:
I can't even run the latest weekly AppImage. I tried the 1.0.1 AppImage and it also crashed when clicking Apply

maxwxyz, 2025-08-13:
in safe mode?

ryankembrey, 2025-08-13:
Actually, 1.0.1 AppImage doesnt crash in safe mode. But it does crash when not in safe mode