FreeCAD/FreeCAD #23078: Assembly: Not possible to edit joint references

maxwxyz, 2025-08-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There is no way of changing reference faces or elements of an existing joint in the task dialog. Once the elements are selected, I cannot replace or remove them. The only way is to delete the joint and recreate it from scratch:

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods:
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-08-13:
Yes it was possible to remove by either clicking again on the selected element to unselect it (which is often not possible due to accessibility of the element)
Or option 2 select in the list and press delete.

Can you confirm which is broken? Both?

drwho495, 2025-08-13:
> Yes, both. Normally you also have a right click delete in such dialogs which is also not working.

could you apply and reopen the joint? i find that its only broken when you first select the faces to joint

closed 2025-08-25 by chennes