FreeCAD/FreeCAD #23081: Draft/BIM: The search for the snap point is not correct.

kaiwas, 2025-08-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In short, the snap should search within a certain radius ON THE SCREEN see the picture),  not the object.

[image]

That is, the closer we get to the point, the more accurate and precise the binding.
Now, very often, the snap point flies off the screen when zoomed in too much.

In the video, you can see a simple example in which I am building a polyline in a rectangle (to fill it).. But despite the fact that the cursor is located very close to the vertex that I need (and even a snap icon appears), the snap occurs to a completely different point, which is far away.

This also applies to other types of snap (for example, the middle).
I understand that the line to which the snap is made is highlighted, but catching the moment when the right line for the correct snap is highlighted can be very difficult.

As a result, working with snap turns into some kind of constant suffering and attempts to catch the right point. Rollbacks and new and new attempts to do everything correctly. The feeling is that the snap does not help, on the contrary, it makes the task more difficult.

When the object is far away and there are many points concentrated in one place, we still have to zoom in quite a bit.
But if in other programs this gives a useful result, then in FreeCAD it only makes the situation worse, because it is easy to snap by mistake (it will simply jump off the screen)

https://github.com/user-attachments/assets/82cd2d66-cf43-4c0e-83c5-99f76e1107f6

Test file. Remove the zip extension.

[Snapsss.FCStd.zip](https://github.com/user-attachments/files/21752495/Snapsss.FCStd.zip)

A small example from Corel Draw (Note that this is not CAD).
The snap works by radius on the screen and does not stick to the beginning or end if they are located outside the visibility (or even outside the radius). I don't even know what radius this is and I can't change the setting, BUT I am always sure that the snap will work as it should.

https://github.com/user-attachments/assets/ba56f463-a064-4c3a-9f29-838ba298015c

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git) AppImage
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Curves.backup1751366017.4487264 (Disabled)
  * addFC-main 1.6.2
  * Reinforcement
  * Curves.backup1751365993.7542331 0.6.63 (Disabled)
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * FrameForge 0.1.3
  * Curves.backup1751366131.5540717 (Disabled)
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.64
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-13:
@Roy-043 FYI

semhustej, 2025-08-14:
This is a good observation! I never realized this even though instinctively the snapping sometimes does not feel right.

It seems that in AutoCAD for instance the closeness of the endpoint (or other important points) overrides the snap of the top most object. This leads to a much **more predictable snapping experience**, as when I hover my mouse around the point, I usually want to snap the point.

I think in FreeCAD this would mean that if a snap point is close enough to the cursor, the snap to the top most object should be overridden by the snap to the close snap point.

I tried to make the point with this picture:

[image]

I am not sure this should happen with all snap points. It should definitely exclude 'snap near'

Let's see what Roy thinks about this issue and than we can think of other snap points that should be excluded from this behavior. I think if this was