FreeCAD/FreeCAD #23083: FEM: Values need to be unique error when running frequency analysis without boundary conditions

FEA-eng, 2025-08-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've noticed that sometimes when running frequency analyses without boundary conditions, the old ccx solver (I can't reproduce it with the refactored one even though IMO there should be no difference in this regard) throws this error:

```
14:17:03  loading results failed
14:17:03  Values need to be unique
```

or more when submitting with FEM_SolverRun:

```
14:27:57  CalculiX read results...
14:27:57  Read ccx results from frd file: C:\Users\jakub\AppData\Local\Temp\fcfem_w8_j2afk\FEMMeshNetgen.frd
14:27:57  Running the Python command 'FEM_SolverRun' failed:
Traceback (most recent call last):
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\femcommands\commands.py", line 1183, in Activated
    run_fem_solver(self.selobj)
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\femsolver\run.py", line 110, in run_fem_solver
    fea.run()  # standard, no working dir is given in solver
    ^^^^^^^^^
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 638, in run
    self.load_results()
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 756, in load_results
    self.load_results_ccxfrd()
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 766, in load_results_ccxfrd
    importCcxFrdResults.importFrd(
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\feminout\importCcxFrdResults.py", line 251, in importFrd
    setupPipeline(
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\feminout\importCcxFrdResults.py", line 74, in setupPipeline
    pipeline_obj = ObjectsFem.makePostVtkResult(doc, result_data, results_name)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\FreeCAD_weekly-2025.08.13-Windows-x86_64-py311\Mod\Fem\ObjectsFem.py", line 681, in makePostVtkResult
    obj.load(*result_data)

Values need to be unique
```

and displays no frames. There's also the "Unable to import data extraction widget" error when opening the pipeline but it seems to be more related to #23030.

I attached a test file.

[Unique_values_error.FCStd.zip](https://github.com/user-attachments/files/21754458/Unique_values_error.FCStd.zip)

@ickby and @marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-08-13:
You won't necessarily get similar behavior between the two solvers.
The new solver uses a completely different function to read results from the .frd file generated by CalculiX.
For example, the old solver doesn't correctly load heat flux values for 2D thermomechanical analysis, while the new one does.

FEA-eng, 2026-02-20:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-08-31:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

td65-cmd, 2026-10-30:
Hi,
I'm struggling with similar issue on 1.1.4 version.
Used to perform 2D axis FEM but recently none of them are working propely even on very simple case.
I've loaded old fils and relaunchng the solver works properly but once mesh is regenerated, I'm facing the same issue.
Tried on portable version of freecad with same behavior :
08:49:52  loading results failed
08:49:52  Values need to be uniquepy
Exception: Traceback (most recent call last):
  File "C:\...\FreeCAD_1.1.4-Windows-x86_64-py311\Mod\Fem\femguiutils\data_extraction.py", line 86, in __init__
    self.onPostDataChanged(self.Object)
  File "C:\...\Logiciel\FreeCAD_1.1.4-Windows-x86_64-py311\Mod\Fem\femguiutils\data_extraction.py", line 145, in onPostDataChanged
    points = algo.GetOutputDataObject(0).GetPoints().GetData()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'GetData'
08:53:22  Unable to import data extraction widget

I've tried simple case on tensile test with same issues :
09:05:48  loading results failed
09:05:48  Values need to be uniqueActive Task Dialog found! Please close this one before opening  a new one!
09:07:07  pyException: Traceback (most recent call last):
  File "C:\...\AppData\Local\Programs\FreeCAD 1.1\Mod\Fem\femguiutils\data_extraction.py", line 86, in __init__
    self.onPostDataChanged(self.Object)
  File "C:\...\AppData\Local\Programs\FreeCAD 1.1\Mod\Fem\femguiutils\data_extraction.py", line 145, in onPostDataChanged
    points = algo.GetOutputDataObject(0).GetPoints().GetData()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'GetData'
09:07:07  Unable to import data extraction widget
09:07:20  pyException: Traceback (most recent call last):
  File "C:\...\AppData\Local\Programs\FreeCAD 1.1\Mod\Fem\femguiutils\data_extraction.py", line 86, in __init__
    self.onPostDataChanged(self.Object)
  File "C:\...\AppData\Local\Programs\FreeCAD 1.1\Mod\Fem\femguiutils\data_extraction.py", line 145, in onPostDataChanged
    points = algo.GetOutputDataObject(0).GetPoints().GetData()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'GetData'
09:07:20  Unable to import data extraction widget

But at the end data visualisation is ok on 3D model ... not on 2D.

Any solution on that ?

ickby, 2026-11-03:
> 17\. Hi,
>      I'm struggling with similar issue on 1.1.4 version.
>      Used to perform 2D axis FEM but recently none of them are working propely even on very simple case.
>      I've loaded old fils and relaunchng the solver works properly but once mesh is regenerated, I'm facing the same issue.
>      Tried on portable version of freecad with same behavior :
>      08:49:52  loading results failed
>      But at the end data visualisation is ok on 3D model ... not on 2D.
>      Any solution on that ?

I think this sounds like a different issue. I can look into it if you can provide the file. Could you open a new issue for that, with the file attached? This would be very helpful.

td65-cmd, 2026-11-04:
>      I think this sounds like a different issue. I can look into it if you can provide the file. Could you open a new issue for that, with the file attached? This would be very helpful.

First of all, thanks for the feedback (my first time here in Github). I've found a way around to overcome this issue : saving file with results and reopen it. It works so far. What files would you need to investigate that (.frd are massive) ? 

Uploaded screenshots from results (end of ccx run on left side and reloaded results on left side)

[image]

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/e0f94fcb60400281e2ec75482605f8fd275d5bd6

Close frequencies are kept at full precision instead of rounding to the same frame value. The separate 2D data-extraction issue mentioned above isn't covered.