FreeCAD/FreeCAD #23087: BIM: Create Arch_Space when several solids are selected

semhustej, 2025-08-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Feature request:
Priority: low

**Current state:**
Standard `Arch_Space` creation works only with one solid (or another object) selected. When two solids are selected and `Arch_Space` is invoked, the result is an empty space object created (where user is only informed if `Report view` is open - 'Arch: error computing space boundary for Space').

See the behavior in the video:

https://github.com/user-attachments/assets/dd1361fc-8590-4522-a3d8-294c7fa38999

**Typical workflow description:**
Space is most commonly used for creation of Rooms in FreeCAD. The typical workflow would be to create solid object for each room and than create Space out of it. If user could create several space objects with one command, this would speed up this typical workflow quite a bit.

**Proposed behavior:**
When several objects are selected and `Arch_Space` command invoked, the command should create independent space based on each of the selected objects. 
Similar way as `Arch_Pipe` works for instance.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-08-15:
Note: it should be mentioned that the requested behavior should not conflict with the option to select several faces and create new space object based on these faces as a bounding objects.

paullee0, 2025-08-16:
As 2 parallel faces are selected, no valid volume could be created supposedly.  Maybe the command should return warning like above for this case?

Have tried to select the 2 objects in the tree for avoidance of doubt and invoke the command?

semhustej, 2025-08-16:
> 
> Have tried to select the 2 objects in the tree for avoidance of doubt and invoke the command?

I tested this and the result is the same as when selecting the 2 objects (faces) from the 3D view.

paullee0, 2025-08-16:
After checking the wiki and codes, either cases are not legitimate in creating a Space, so the return error message is valid.

More clear error is helpful though.

semhustej, 2025-08-16:
> After checking the wiki and codes, either cases are not legitimate in creating a Space, so the return error message is valid.
> 
> More clear error is helpful though.

Thanks for looking into the issue. So no way of creating several space objects using the command only once? It's a shame but not a deal breaker. You don't create space objects every day on a project.

If that is the case, I think this kind of notification would be more informative: `Only one Space object at a time can be created. Creating empty Space object.`

semhustej, 2025-08-16:
> 
> But then, the question is what are valid combinations of objects, subelements and the desirable / intuitive results
> 
I see that when 

- faces, edges or vertecies from multiple objects are selected, the tool tries to create space based on the envelope that these are creating
- several faces, edges or vertecies from single object are selected, the tool tries to create space based on the envelope that these are creating
- single face, edge or vertex from an object are selected, the tool creates space from the object

That would leave an option:

- when several whole solid objects are selected (it does not matter whether from project tree or 3D space), the space tool should attempt to create several spaces  - one from each object.

Now if that would be difficult to code, I don't think it's worth currently spending time on it. As stated above, I think this one has a pretty low priority. When reporting the request, I completely forgot about the option of creating the space from bounding faces.

> But I do not find the concept of 'boundaries' works well at the moment - discussed earlier somewhere else like 'concave' arrangement not working to my memory.
> 

Yeah, I remember discussion about that. But the tool works well with simple boundaries.

paullee0, 2025-08-17:
Indeed, I do not understand the logic when there is multiple selection at the moment :)

Playing around boundaries, it seems the task panel is more intuitive.

[image]

[image]

[image]