FreeCAD/FreeCAD #23113: TechDraw dimensions wrong

Beep6581, 2025-08-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a geometry that is 970 mm long, but TechDraw dimension shows it as 883.29 mm.

[sandblaster.FCStd.zip](https://github.com/user-attachments/files/21779232/sandblaster.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42934 (Git)
Build date: 2025/08/13 20:46:20
Build type: Release
Branch: main
Hash: 81d22eb2c338d89ea85848f31bfc0cde20d4c066
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.39
  * CurvedShapes 1.0.13
  * woodworking 1.0.1.39285.1.1.1.2025.08.08
  * Curves
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Beep6581, 2025-08-14:
I couldn't get `TechDraw_DimensionRepair` to work when selecting edges in TechDraw view, but it worked when selecting them in 3D view. Bug or not, I don't know.

Confirmed that this issue stems from the foreshortened isometric view.

Would be nice if  `TechDraw_AxoLengthDimension` was incorporated into the `TechDraw_Dimension` tool, so that the switch to the correct tool happens automatically when working with a isometric view.

WandererFan, 2025-08-15:
I don't really like TechDraw_AxoLengthDimension as it is a lot of selecting and I suspect you can create a incorrect measurement if you pick the wrong edges.   Works when you use it correctly though.

Without using TechDraw_AxoLengthDimension, you can do this:
[image]

[sandblaster_3drefs.FCStd.not.zip](https://github.com/user-attachments/files/21782251/sandblaster_3drefs.FCStd.not.zip)

Beep6581, 2025-08-16:
How much work was it to get a single dimension correct?