FreeCAD/FreeCAD #23118: Sketcher: Geometry jumps around if editing constraint by a too large amount

matthiasdanner, 2025-08-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I observed a strange behaviour when editing constraint values. When changing the value only a bit the sketch moves as expected, but when changing the value by a larger amount (in my case from 20mm to 100mm or more) the geometry suddenly is below the x-Axis. 

Also tested in Safe mode, there was the same behaviour.

Testfile: [Unbenannt.FCStd.zip](https://github.com/user-attachments/files/21780660/Unbenannt.FCStd.zip)

Screencast: https://github.com/user-attachments/assets/377f4445-9c0f-439c-936b-6cfd984da9a6

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42859 (Git)
Build date: 2025/08/05 20:55:16
Build type: Release
Branch: sketcher-constraint-group-selection
Hash: 5b500176b2fdc6294c6d17e2a20ec45db345121f
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
```

### Subproject(s) affected?

Sketcher

### Anything else?

n/a

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-08-14:
This is the result of solvers being OK with outright flipping the orientation of constraints when converging towards new solutions. As [the wiki mentions](https://wiki.freecad.org/Sketcher_Workbench#Constraints),

> A Sketch may have a number of different solutions, and making it fully-constrained can mean that just one of these possible solutions has been reached based on the applied constraints. See also: [Flipping](https://wiki.freecad.org/Sketcher_Workbench#Flipping).

I also think this situation is terrible both mathematically and in terms of user experience, egregiously so as flipping the solution back by hand is *tedious* at best.

Probably a duplicate of #17579. Related: #20898.

matthiasdanner, 2025-08-15:
Oh thanks for pointing that out. I'll close this issue then

closed 2025-08-15 by matthiasdanner