FreeCAD/FreeCAD #23124: UI: "Select module" dialog shows untranslated title and importer names

ElementW, 2025-08-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The "Select module" window:

* Has an untranslated window title.
  Broken by a capitalisation change by @maxwxyz in ecf02b787814c1020654c74606f19bb3e06b273e
* Shows untranslated importer file type names.
  `FileDialog` relies on parsing file match strings (e.g. `SVG as geometry (*.svg *.SVG)`) which is notoriously error prone and rarely translated

[image]

### Full version info

```shell
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git) AppImage
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyl/media/source/FreeCAD/Mod/Mesh/Gui/Command.cppe: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.38
  * FreeCAD-Ribbon 1.9.0.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-15:
@ElementW is there an issue with the dialog or import or is it just not translated?

ElementW, 2025-08-15:
@maxwxyz Just lack of translation.

In the case of the line you changed, the translation files didn't follow; not blaming you for it it's just an honest mistake that ends up decreasing the i18n support in status quo FreeCAD.

For the importer names/actions, it's because they rely on the same file filter strings that are used in the system-provided "Import" file dialog, in said dialog it's not as bad because the extensions are shown, but here there's only the name. No file match string is translated or translatable (lacking `tr()`/`translate()`) in the entirety of FC except for the Arch workbench's `IFC files (*.ifc)` and `Image file (*.png *.jpg *.bmp)` strings.

ElementW, 2025-08-15:
@maxwxyz In fact in updating those strings you broke most translations even though you only changed the case. A find&replace on the translation XML files is probably also in order.

E.g. in these screenshots every English string is one you changed with your commit series.
[image]

[image]

maxwxyz, 2025-08-15:
Yes, the new strings are not translated yet. IIRC new guidelines for each language are also in the making so the translations are also more consistent.

maxwxyz, 2025-08-15:
The changes did not make strings untranslatable or added new strings. Only the text itself was adjusted.

ElementW, 2025-08-15:
I know, they're unrelated to your changes, but there's 2 bullet points on my issue description, and this is about the latter. That strings are untranslated at any point is not an *issue per se*, pending translation through Crowdin, but the code preventing those translations for certain strings altogether is a bug in my opinion.

luzpaz, 2025-10-02:
@maxwxyz to late to get a fix for this into 1.1 ?