FreeCAD/FreeCAD #23134: CAM: Pocket G3 calculation error

dimichelec, 2025-08-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On a part design having 4 symmetric pockets, only one produces an erroneous gcode result.

[image]

The pocket does not cut all the way thru the depth of the block, leaving just a 0.5mm at the base.

[image]

I'm creating a pocket operation for each of the 4 highlighted pockets and all but the bottom-left (closest to origin), result in appropriate grbl gcode. The incorrect one results in something like this (note: `G3 X4.585 Y1.922 Z-0.250 I-8746784.696 J1.698 K0.000 F200.000` below:

```
(Exported by FreeCAD)
(Post Processor: grbl_post)
(Output Time:2025-08-15 16:40:30.478186)
(Begin preamble)
G17 G90
G21
(Begin operation: Fixture)
(Path: Fixture)
G54
(Finish operation: Fixture)
(Begin operation: 1/8" endmill001)
(Path: 1/8" endmill001)
(1/8" endmill001)
(Begin toolchange)
( M6 T2 )
M3 S10000
(Finish operation: 1/8" endmill001)
(Begin operation: top_slot1)
(Path: top_slot1)
(top_slot1)
G0 Z5.000
G0 X5.300 Y2.286
G0 Z1.000
G1 X5.300 Y2.286 Z-0.250 F20.000
G2 X5.415 Y1.922 Z-0.250 I-0.293 J-0.293 K0.000 F200.000
G1 X5.415 Y-0.415 Z-0.250 F200.000
G1 X4.585 Y-0.415 Z-0.250 F200.000
G3 X4.585 Y1.922 Z-0.250 I-8746784.696 J1.698 K0.000 F200.000
G1 X4.600 Y2.079 Z-0.250 F200.000
G2 X5.300 Y2.286 Z-0.250 I0.406 J-0.087 K0.000 F200.000
G1 X5.300 Y2.286 Z-0.500 F20.000
...
```

Any of the others are fine, for example:

```
(Exported by FreeCAD)
(Post Processor: grbl_post)
(Output Time:2025-08-15 16:41:26.404723)
(Begin preamble)
G17 G90
G21
(Begin operation: Fixture)
(Path: Fixture)
G54
(Finish operation: Fixture)
(Begin operation: 1/8" endmill001)
(Path: 1/8" endmill001)
(1/8" endmill001)
(Begin toolchange)
( M6 T2 )
M3 S10000
(Finish operation: 1/8" endmill001)
(Begin operation: top_slot2)
(Path: top_slot2)
(top_slot2)
G0 Z5.000
G0 X45.300 Y2.286
G0 Z1.000
G1 X45.300 Y2.286 Z-0.250 F20.000
G2 X45.415 Y1.922 Z-0.250 I-0.293 J-0.293 K0.000 F200.000
G1 X45.415 Y-0.415 Z-0.250 F200.000
G1 X44.585 Y-0.415 Z-0.250 F200.000
G1 X44.585 Y1.922 Z-0.250 F200.000
G1 X44.600 Y2.079 Z-0.250 F200.000
G2 X45.300 Y2.286 Z-0.250 I0.406 J-0.087 K0.000 F200.000
G1 X45.300 Y2.286 Z-0.500 F20.000
G2 X45.415 Y1.922 Z-0.500 I-0.293 J-0.293 K0.000 F200.000
G1 X45.415 Y-0.415 Z-0.500 F200.000
G1 X44.585 Y-0.415 Z-0.500 F200.000
G1 X44.585 Y1.922 Z-0.500 F200.000
G1 X44.600 Y2.079 Z-0.500 F200.000
G2 X45.300 Y2.286 Z-0.500 I0.406 J-0.087 K0.000 F200.000
G1 X45.300 Y2.286 Z-0.750 F20.000
G2 X45.415 Y1.922 Z-0.750 I-0.293 J-0.293 K0.000 F200.000
G1 X45.415 Y-0.415 Z-0.750 F200.000
...
```

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.41
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dimichelec, 2025-08-15:
[x-axis block top.zip](https://github.com/user-attachments/files/21803883/x-axis.block.top.zip)

tarman3, 2025-08-15:
I confirm
And see problem in all Pocket operations
The path is distorted

Dimitris75, 2025-08-17:
I completely eliminated the If statement, but I am still experiencing similar outcomes.

I'm receiving an extra GeomAbs_Line . Part of the Arc is a straight line now and the Arc is smaller than it should.

https://github.com/FreeCAD/FreeCAD/blob/1a4ff21bdda4eabadc90415053e486e7c2381415/src/Mod/CAM/App/Area.cpp#L437-L444

Someone must conduct a deeper investigation...