FreeCAD/FreeCAD #23136: Freecad 1.0

OrtieD, 2025-08-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Doesn't work on macOS Tahoe Open Beta 3

### Full version info

```shell
n thrown (Unable to find zip structure: End-of-central-directory)
10:04:32  C++ exception thrown (Unable to find zip structure: End-of-central-directory)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-16:
Do the weekly builds work: https://github.com/FreeCAD/FreeCAD/releases/ ?
@chennes @hyarion do you have macs with the latest OS beta?

hyarion, 2025-08-16:
> @hyarion do you have macs with the latest OS beta?

Nope, I always wait about a version to make sure all tools I use work before I switch 🤓

luzpaz, 2025-08-16:
@JohnOCFII have you seen this before on macOS ?

OrtieD, 2025-08-16:
> Do the weekly builds work: https://github.com/FreeCAD/FreeCAD/releases/ ? [@chennes](https://github.com/chennes) [@hyarion](https://github.com/hyarion) do you have macs with the latest OS beta?

I'm using latest weekly release

closed 2025-08-16 by OrtieD

luzpaz, 2025-08-16:
@OrtieD It works for you on weekly ? (that's why you closed it?) 
(We may need to keep this open for backporting a fix to 1.0.x series)

OrtieD, 2025-08-16:
sorry, it doesn't work, keep it open

luzpaz, 2025-08-16:
@OrtieD can you please add the full [About](https://wiki.freecad.org/About) info for each version you tested on ? TIA

JohnOCFII, 2025-08-16:
> [@JohnOCFII](https://github.com/JohnOCFII) have you seen this before on macOS ?

I haven't seen this issue, but then, I'm not running the Tahoe OS Beta.

```
OS: macOS 15.6
Architecture: arm64
Version: 1.1.0dev.42942 (Git)
Build date: 2025/08/15 15:09:23
Build type: Unknown
Branch: main
Hash: af83b6883eb0afae7d82aaaefc6a3991429c0a70
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.8.15
  * manifest.json
  * Telemetry 1.0.2
```

lagnat, 2025-08-17:
Weekly is working fine on Tahoe 26.0 Beta (25A5338b)

luzpaz, 2025-08-17:
@OrtieD run `FreeCAD--version` from the CLI

luzpaz, 2025-08-17:
@lagnat can you re-edit your post and add your full About info ? TIA

OrtieD, 2025-08-19:
Just installed TAHOE 26 beta 4 and both, Freecad 1.02 and weekly release (25.08.13) are working fine!

luzpaz, 2025-12-07:
Related ?
* https://github.com/FreeCAD/FreeCAD/issues/18536