FreeCAD/FreeCAD #23144: QImageIOHandler error in the Start tab

carribeiro, 2025-08-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The problem started right after I started FreeCAD, no action necessary. The message goes like this, and is repeated from time to time.

10:12:34  (qt.gui.imageio) QImageIOHandler: Rejecting image as it exceeds the current allocation limit of 256 megabytes

The message stops after closing the Start tab. It must be some of the images that's being loaded there.

It's not a critical error but it's annoying as it clutters the report vew and nags the user from time to time with an alert box.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
Installed mods: 
  * Curves 0.6.62
  * fasteners 0.5.38
  * FreeGrid 2.2.0
  * LCInterlocking
  * Manipulator 1.6.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rudiloos, 2025-08-17:
I am experiencing exactly the same.

**Version Info**
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/South Africa (en_ZA)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * fasteners 0.5.38
  * kicadStepUpMod 11.7.1
  * LCInterlocking
  * sheetmetal 0.7.24
```

luzpaz, 2025-12-15:
@hds4 please open a new ticket to avoid a discussion starting on this solved one.

tetektoza, 2025-12-15:
@hds4 yeah, I think it's a good idea, you can put that into the issue. I may be faster and open PR soon.