FreeCAD/FreeCAD #23162: PartDesign: chamfer causes part of surface to disappear

brettowe, 2025-08-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

adding a chamfer to the hole 
[image]

causes part of the surface to disappear
[image]

[stihl-a2.FCStd.zip](https://github.com/user-attachments/files/21823362/stihl-a2.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.42942 (Git) AppImage
Build date: 2025/08/15 15:09:23
Build type: Release
Branch: main
Hash: af83b6883eb0afae7d82aaaefc6a3991429c0a70
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.8077/1
Installed mods: 
  * ThreadProfile 1.96.0
  * lattice2 1.0.0
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

brettowe, 2025-08-17:
i ended up using a fillet as a workaround.
I'll note since i forgot to mention that 0.5mm was used on the chamfer to cause the issue, using smaller sizes did avoid it

maxwxyz, 2025-08-31:
I'll close this issue as there are several chamfer/fillet related OCC issues

closed 2025-08-31 by maxwxyz