FreeCAD/FreeCAD #23204: Need to add "btl-notes" to property bag for a tool bit

ShamanTcler, 2025-08-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Error when loading a tool bit.

08:14:24    File "C:\FreeCadBeta\Mod\CAM\Path\Tool\library\ui\dock.py", line 165, in _add_tool_controller_to_doc
08:14:24      selected_tools = self._add_tool_to_doc()
08:14:24                       ^^^^^^^^^^^^^^^^^^^^^^^
08:14:24    File "C:\FreeCadBeta\Mod\CAM\Path\Tool\library\ui\dock.py", line 132, in _add_tool_to_doc
08:14:24      toolbit.attach_to_doc(FreeCAD.ActiveDocument)
08:14:24    File "C:\FreeCadBeta\Mod\CAM\Path\Tool\toolbit\models\base.py", line 485, in attach_to_doc
08:14:24      self.attach_to_obj(tool_doc_obj, label=label)
08:14:24    File "C:\FreeCadBeta\Mod\CAM\Path\Tool\toolbit\models\base.py", line 513, in attach_to_obj
08:14:24      temp_obj.copy_to(self.obj)
08:14:24    File "C:\FreeCadBeta\Mod\CAM\Path\Tool\docobject\models\docobject.py", line 181, in copy_to
08:14:24      obj.addProperty(prop_type, prop_name, prop_group, prop_doc)
08:14:24  **NameError: Invalid property name 'btl-notes'**

### Full version info

```shell
OS: Windows 11 build 26120
Architecture: x86_64
Version: 1.1.0dev.42975 (Git)
Build date: 2025/08/18 17:31:10
Build type: Release
Branch: main
Hash: bafd9f75b50633cf533e13e55b00116be53adc2b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * A2plus.zip
  * btl.zip
  * Curves 0.6.63
  * lattice2 1.0.0
  * woodworking 1.1.0.dev.1.1.1.2025.08.14
```

### Subproject(s) affected?

None

### Anything else?

Should be an easy fix. I would be interested in looking at the fix so in the future I would be able to submit the fix myself.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-19:
@ShamanTcler can you test in Safe Mode ?

chennes, 2025-08-19:
The "installed mods" list is just an `ls` of the Mod directory.

luzpaz, 2025-08-19:
@chennes a2plus for sure has a `package.xml`[^1] file. Any idea why it isn't pulling the version info from it ?

[^1]: https://github.com/kbwbe/A2plus/blob/master/package.xml