FreeCAD/FreeCAD #23221: Shape distorts when redimensioning

delhub, 2025-08-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When redimensioning a shape while in Sketcher, the image gets distorted.  I can share a screen capture showing what happens during this process, if that's helpful.  I am following a tutorial at: https://www.youtube.com/watch?v=NpJ-_QtdBwM&list=PLEBQazB0HUyTQkRkbD02DZqnlV6oBMhHB&index=8.  At the point where there is only one DOF left to resolve (distance from the origin to the shape), here is what's shown in the video:

1. Resolve the final distance by setting the dimension to a value.
2. Redimension the value to a different distance (say, 70 mm) - issue may or may not happen.
3. Redimension the value to a different distance (say, 100 mm) - issue may or may not happen.
4. Redimension the value back to 70 mm - issue occurs.  Did this three times.

This is what's shown in the Report view:

11:48:09  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Invalid constraint index: 10
11:48:09  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(10,App.Units.Quantity('70.000000 mm'))

===  Macro Recording ===

# -*- coding: utf-8 -*-

# Macro Begin: C:\Users\Tony\AppData\Roaming\FreeCAD\Macro\RedimensioningError.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import Sketcher

# Gui.runCommand('Std_DlgMacroRecord',0)
# Gui.Selection.addSelection('Unnamed','Body','Sketch.')
# Gui.Selection.clearSelection()
# Gui.runCommand('Sketcher_CompDimensionTools',0)
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('Distance',3,1,-1,1,12.263835)) 
# Gui.Selection.addSelection('Unnamed','Body','Sketch.Vertex8',-12.2638,0,0)
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('DistanceX',3,1,-1,1,12.263835)) 
App.getDocument('Unnamed').getObject('Sketch').setDatum(16,App.Units.Quantity('12.260000 mm'))
App.ActiveDocument.recompute()
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Body','Sketch.Constraint17',-6.31663,-0.00899965,-2.94192,False)
App.getDocument('Unnamed').getObject('Sketch').setDatum(16,App.Units.Quantity('70.000000 mm'))
App.ActiveDocument.recompute()
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Body','Sketch.Constraint17',-37.262,-0.00899963,-3.08158,False)
App.getDocument('Unnamed').getObject('Sketch').setDatum(16,App.Units.Quantity('100.000000 mm'))
App.ActiveDocument.recompute()
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Body','Sketch.Constraint17',-53.1775,-0.0089998,-1.65868,False)
App.getDocument('Unnamed').getObject('Sketch').setDatum(16,App.Units.Quantity('70.000000 mm'))
App.ActiveDocument.recompute()
# Gui.Selection.clearSelection()
# Macro End: C:\Users\Tony\AppData\Roaming\FreeCAD\Macro\RedimensioningError.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

[RedimensionErrorFile.zip](https://github.com/user-attachments/files/21860081/RedimensionErrorFile.zip)
[redimensioningError_mp4.zip](https://github.com/user-attachments/files/21860080/redimensioningError_mp4.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-22:
@Roy-043 is there something we can do via the UI about this or should we close ticket ?

Roy-043, 2025-08-24:
The error message points to a different problem than flipping. I have not been able to trigger that error.

delhub, 2025-08-24:
I'd be glad to assist in any way I can to provide any additional information you may need.

luzpaz, 2025-11-17:
I compressed the clip down so it's viewable upon opening the ticket. 

@delhub any chance you can test on 1.2 weekly development build  ?  https://github.com/FreeCAD/FreeCAD/releases