FreeCAD/FreeCAD #23225: Cannot go to linked assembly in Assembly3 workbench

MisterMakerNL, 2025-08-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the Assembly3 workbench, I cannot go to the linked part if the "part" itself is an assembly. This impacts navigation and workflow efficiency when working with nested assemblies.
Part body:
[image]

Assembly:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42953 +10 (Git)
Build date: 2025/08/18 11:15:46
Build type: Release
Branch: Version_specific_settings
Hash: 4af49c0c511e2e0affad1c44bd95859bf1ffbda5
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD_Stylesheet_roaming.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/69.8349/2
Installed mods: 
  * 3D_Printing_Tools
  * addFC 1.8.5
  * AddonManager 2025.8.16
  * Assembly3 0.12.2
  * BIM021 2021.12.0
  * Curves 0.6.69
  * Dracula 0.0.9
  * fasteners 0.5.41
  * FeedsAndSpeeds 0.6.0
  * freecad_metal_workbench 0.0.1
  * Gridfinity 0.11.9
  * Iconpack 2023.10.25
  * IconThemes (Disabled)
  * lattice2 1.0.0
  * manifest.json
  * OpenTheme 2025.5.20
  * Render 2024.12.15
  * scanObjects
  * sheetmetal 0.7.24
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2
  * woodworking 1.1.0.dev.1.1.1.2025.08.16
```

### Subproject(s) affected?

Assembly

### Anything else?

Assembly3 version: 0.12.2. No related issues found.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-20:
So this is a feature request for the Assembly3 addon?

maxwxyz, 2025-08-21:
You mention Assembly3 everywhere, could you rephrase the issue and reopen?

MisterMakerNL, 2025-08-21:
> You mention Assembly3 everywhere, could you rephrase the issue and reopen?

All those different assembly workbenches are confusing. I removed the 3.

PaddleStroke, 2025-11-21:
Done. Requires https://github.com/FreeCAD/FreeCAD/pull/24623 to merge before I can open PR.