FreeCAD/FreeCAD #23243: Start page drop-down list of languages is too long

FEA-eng, 2025-08-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This list is way too long and spans through my entire screen. Maybe splitting it into two rows would make sense:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mpfusion, 2025-08-20:
> IMO yes. My locale is Polish but I always use FreeCAD in English

Which totally makes sense if the translations are incomplete or awkward. As I mentioned, this doesn't require a long drop-down menu on the start page. That's easy to set on a per-program basis. E.g. start freecad with `LANG=en_US freecad`

obelisk79, 2025-08-20:
I wonder if this is related to the fusion QStyle

Syres916, 2025-08-20:
https://stackoverflow.com/questions/52778846/how-to-change-qcombobox-items-height

FEA-eng, 2025-08-20:
> > IMO yes. My locale is Polish but I always use FreeCAD in English
> 
> Which totally makes sense if the translations are incomplete or awkward. As I mentioned, this doesn't require a long drop-down menu on the start page. That's easy to set on a per-program basis. E.g. start freecad with `LANG=en_US freecad`
> 
> And there's also a setting in the Preferences.

Ok, but the whole idea of this First start page is that it provides main settings that you may want to change right away after starting FreeCAD. I think that language should be one of them. Same with theme, unit system or navigation style. The rest can be customized later in the Preferences or in other ways.

However, maybe the list could be limited to the current locale and English? A full list with proper sizing might be better though.

I guess the easiest way would be to shorten it and add auto scrolling. For example, here's Gimp's welcome screen:

[image]

And its languages drop-down:

[image]

The top option is "System language".

Reqrefusion, 2025-08-20:
I think there should be three options when you click the language selection: Show the language used on the computer, English, and Show others. I don't think a user would choose any other language; they'd probably switch languages to English anyway. That's why it's best for other languages to remain hidden until the "Show others" button is pressed.