FreeCAD/FreeCAD #23250: Core: Exception messages are not useful.

leoheck, 2025-08-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am getting tons of messages like this when loading existing designs using the latest AppImage.

```
13:01:09  <Exception> No axis reference specified
13:01:09  <Exception> No mirror plane reference specified
13:01:42  <Exception> No direction reference specified
13:01:42  <Exception> No mirror plane reference specified
```

The Exception messages are generic and not useful at all.

How do I get info on which objects each of those messages is related to so I can track and fix the problem?

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42975 (Git)
Build date: 2025/08/18 17:31:10
Build type: Release
Branch: main
Hash: bafd9f75b50633cf533e13e55b00116be53adc2b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * manifest.json
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2-dev
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * osh-autodoc-workbench 0.2.3
  * FreeCAD-DFM-Workbench
  * Assembly4p1 0.50.18
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.3
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-08-20:
> Anyways, the exceptions should have more info. It should have the name of the related object/feature on them. Otherwise there is no way of finding these objects on big projects/designs with multiple parts.

I'd agree -- but I think it might be a side effect of the recent transparency preview work.  I believe @kadet1090 was looking into it.

leoheck, 2025-09-09:
Still getting this on Polar Pattern using `1.1.0dev.14555 (Git shallow)| 2025/09/02 21:58:19`

```
01:37:43  <Exception> No axis reference specified
```

I am not sure if it was fixed. 

The message is generated when creating the latest polar pattern here [polar_pattern_issue.FCStd.zip](https://github.com/user-attachments/files/22226306/polar_pattern_issue.FCStd.zip)
To see the message, remove the last feature (Polar Pattern), and add it again.

[image]