FreeCAD/FreeCAD #23257: PartDesign: Can’t thread existing feature

pierreporte, 2025-08-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In FreeCAD, the only way to add a thread is to create a threaded hole using the `PartDesign_Hole` feature. It is not possible to thread an existing hole (say from an imported STEP file) nor to add an external thread to a cylindrical shape. There is a workaround for modeled threads, using boolean operations or helical extrusions but they are not practical and will not work for cosmetic threads. Such standalone feature exist in other CAD programs, see for example the following dialog from Catia:

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * fasteners 0.5.41
  * Curves 0.6.63
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * FusedFilamentDesign 0.25.220
```

### Subproject(s) affected?

PartDesign

### Anything else?

Making a dedicated issue from https://github.com/FreeCAD/FreeCAD/issues/5567 discussion

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pitch-circle, 2025-08-21:
> In FreeCAD, the only way to add a thread is to create a threaded hole using the `PartDesign_Hole` feature.

We have a well-documented Fasteners WB, which contains both screw tap and screw die tools, which will do exactly what you describe (put a female thread into a hole or add a male thread to a cylinder).

Hence the statement above is factually incorrect. Perhaps it would benefit from an edit which correctly states the scope for which it is true (i.e. within PartDesign WB).

pierreporte, 2025-08-21:
Thanks @pitch-circle. I’ve modified the text to take the fasteners workbench into account even though it’s not built-in.

onekk, 2025-10-31:
The point is how you intend to make the non cosmetic threads?

There is no magic in 3d modelling, you have to create a helical path and sweep along it to obtain the thread and eventually subtract the solid to the hole.

Or you have other ideas.

For cosmetic threads it is another different matter, as they could probably done using a texture. (Material WB seems to have lost the main developer and maybe a texture could have been a viable solution).

Kind regards
Carlo D.

pierreporte, 2025-10-31:
No idea about modeled threads, probably the same way threaded holes are done.

For cosmetic threaded holes, there is already PR #22573. It should be merged for 1.2.

onekk, 2025-11-01:
Thanks, it was an interesting reading.
For what it count, I appreciate your work and commitment.

Kind regards
Carlo D.