FreeCAD/FreeCAD #23283: [Regression?] Cut operations no longer preserve intersectional appearance

recursivenomad, 2025-08-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In FreeCAD 0.21.2, I would perform boolean operations, including cuts, on 2 parts in order to deterministically change certain face colours:

[image]

But now in FreeCAD 1.0.2, performing a cut operation reverts to the first-selected part appearance settings:

[image]

It is worth noting that union operations still preserve unique face colours, but cut and intersection do not.

To reproduce:

1) **Open FreeCAD 0.21.2**
2) Create 2 boxes in the part workbench
3) Transform one so they intersect
4) Change the appearance of at least one box
5) Perform a boolean cut operation
6) Observe the colours for each section of geometry being preserved
7) **Open FreeCAD 1.0.2**
2) Create 2 boxes in the part workbench
3) Transform one so they intersect
4) Change the appearance of at least one box
5) Perform a boolean cut operation
6) Observe the intersectional colours being overriden by the first-selected part's colour

Semantically, I can understand why a "cut" would not transfer colour, while a "union" would... However, my argument would be that if colour-transfer is an undesired behaviour, the user can always change the tool colour to match the target object, as opposed to getting rid of this past-feature altogether - as it did offer mild utility.

### Full version info

```shell
OS: Windows 10
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

recursivenomad, 2025-08-21:
Related:
- #11210

drwho495, 2025-08-22:
try in 1.1dev

closed 2025-08-22 by maxwxyz