FreeCAD/FreeCAD #23339: PartDesign: Sketch based on shapebinder not updating after appling binder offset

maczargit, 2025-08-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

External geometry from shapebinder not updating when adding offset to it, after sketch is made.
If offset in shape binder (eg 0.01) is applied before creating sketch and referencing external geometry from it, then changes in offset are correctly applied to sketches.

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/158.196/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-24:
> Build date: 2025/08/12 22:08:35

Can you test on a more recent build please ?

maczargit, 2025-08-24:
Tried on newest weekly build (2025.08.20) - same behavior.

maczargit, 2025-08-30:
[Binder_problem.zip](https://github.com/user-attachments/files/22059924/Binder_problem.zip)

Body001 - sketch is created on binder with 0 offset applied 
Body002 - sketch is created on binder with 1mm offset applied
Both binders have tangent join type selected before creating sketch. Both sketches are based on external geometry edge from corresponding binder.

Now changing Body001 binder offset - nothing happens.
While changing binder offset od Body002 - sketch updates correctly.